libretro/Lakka-LibreELEC #533: Lakka images in ZIP instead of GZ for Windows users

vffhr, 2018-09-06:
Lakka IMG files are available to download in GZ format. Windows by default can't extract GZ files. It can however extract ZIP files. Would it be better to use ZIP files instead of GZ files so that they can be extracted on Windows (and I assume all other platforms) without requiring additional software?

vffhr, 2018-09-17:
Thanks @Ntemis, I had no idea! Now that I look it's right there in the GUI; under "Select image" and hovering over "others", there is "gz" listed.

RobLoach, 2018-09-18:
Would be good to have a link during the install process: https://github.com/lakkatv/lakka-website/issues/24 .

ToKe79, 2018-09-19:
Lakka.tv website now directs the end user to use Etcher, which can handle .img.gz files.

closed 2018-09-19 by ToKe79

RobLoach, 2018-09-19:
Nice work! :+1:  Thanks a lot.

vffhr, 2018-10-08:
@ToKe79 I just noticed that the heading on the [download page](http://www.lakka.tv/get/windows/rpi2/) still says "Download and unzip Lakka", should this be changed to just "Download Lakka" now?

ToKe79, 2018-10-08:
Thank you, I will send PR to correct the heading(s).