libretro/Lakka-LibreELEC #534: Hiding date/time by default

vffhr, 2018-09-06:
When Lakka is newly installed, it will naturally show the incorrect time and date in the top-right corner. I'm not sure if the time and date are automatically set when the device is connected to the internet, but even if this is the case the timezone will probably be incorrect.

It bugs me a little bit that the time is incorrect, and I would rather not go the trouble of setting the timezone just to make the time look right (actually I would rather not see the time at all since I have many other time-telling devices around).

If I'm not the only one who feels like this, then maybe it would be better if the date/time was hidden by default.

ToKe79, 2018-09-16:
The date/time display can be disabled via settings. First check if advanced settings are enabled - Settings -> User Interface -> Show Advanced Settings -> On. Then turn off date/time display - Settings -> User Interface -> Views -> Show date / time -> Off.

Or [set up the time zone manually](https://github.com/libretro/Lakka-LibreELEC/wiki/Timezone-settings).

vffhr, 2018-09-17:
Thanks for the info @ToKe79, I looked and couldn't find this option, I must have had advanced settings off. I still think "off" is a better default but it's nice to know where that setting is.

gouchi, 2018-09-18:
If it is ok for you, we can close this issue ?

closed 2018-09-20 by ToKe79