libretro/Lakka-LibreELEC #537: Rotate Menu for Vertical Arcade Screens

xSPARExSTEWx, 2018-09-11:
Lakka-Generic.x86_64-2.1.1.img with a desktop board
When you change the monitor rotation in the setting it only changes it in the emulation. I know this is not a "bug", but I know a lot of users would like to be able to change the menu rotation so when you are selecting a game you do not have to turn your head sideways to be able to read. I have tired the normal ways you would do this in a normal linux enviroment, but this is so striped down that it is missing and not using the packages to be able to do this. Only way I can think of doing this right now is to change the EDID of the monitor or the BIOS of the graphics card.

gouchi, 2018-09-15:
When t[his issue will be fixed](https://github.com/libretro/RetroArch/issues/6770).

We don't use Xorg on Generic platform so you can't use [xrandr](https://en.wikipedia.org/wiki/RandR).

smokku, 2019-08-25:
I have made a build of Lakka running on X11 (not KMS) that supports screen orientation setting.
https://github.com/smokku/Lakka-LibreELEC/releases

First turn on *Settings > User Interface > Show Advanced Settings*.
Then use *Settings > Video > Screen Orientation* to rotate your whole screen.

closed 2020-08-18 by Ntemis