libretro/Lakka-LibreELEC #539: Raspberry 3b Plus + Lakka 2.1.1 + RGB-PI cable black screen.

Alex9534, 2018-09-13:
- Which version of Lakka are you using? Lakka-RPi2.arm-2.1.1.img
- What system hardware are you using? Raspberry 3b+
- What did you do? I have put rgb-pi.dtbo file inside overlay folder and I have added next lines:

dtoverlay=pwm-2chan,pin=18,func=2,pin2=19,func2=2
dtoverlay=rgb-pi
enable_dpi_lcd=1
display_default_lcd=1
dpi_output_format=6
dpi_group=2
dpi_mode=87
hdmi_timings=1920 1 152 247 280 240 1 3 7 12 0 0 0 60 0 40860000 1

at the end of the config.txt file as indicated here:

[http://www.lakka.tv/doc/CRT-Screens/](url)

[My config.txt file: ](https://forums.libretro.com/uploads/default/original/2X/d/d73985003eebce8b91c5557863043f28a776a721.jpg)

- What did you expect to happen? Lakka starts and is visible on my CRT TV through RGB-PI cable.
- What happened instead? Black screen, nothing is seen on the screen.

kivutar, 2018-12-24:
I now have an RGP-Pi to test this, the issue is that I can't find a CRT with SCART where I'm living.

closed 2020-08-18 by Ntemis

kivutar, 2023-03-26:
According to this we might need to enable some new kernel modules https://www.mortaca.com/rgb-pi/wiki/index.php?title=About_RGB-Pi_and_Raspberry_Pi4

JannoTjarks, 2025-04-19:
Hi! Got these kernel modules enabled? Currently I have also with the latest version of Lakka a blackscreen on my CRT.