libretro/Lakka-LibreELEC #543: Lakka-Generic-i386-2.1.1 not booting

MontyGvMC, 2018-09-14:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? Lakka-Generic-i386-2.1.1 (current stable for PC)
- What system hardware are you using?

I tried to start the live system on different setups
#1: Desktop PC,  Intel i7-4770k, 32 GB RAM, ASUS P8z77-m Mainboard, Geforce GTX 1080, Samsung 850 Evo PRO SSD
#2: Acer Aspire 1551
#3: Acer Aspire One D270

- What did you do?
I downloaded the current stable for PC, put it on an USB drive, booted from the USB

- What did you expect to happen?
Lakka starts

- What happened instead?
It got stuck on the black screen with th Lakka Logo (which I guess is the boot screen?). To make shure it is just starting slowly I waited for 30 min on each system. Nothing changed. Still that screen.

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

I will  provide logs soon following the guide (hadn't had the time to play around with ssh last night, i just have windows around so far).

-----
Side question: any way to start Lakka in a maintainence/developer mode enabling logs straigth away and display them on the screen?

gouchi, 2018-09-15:
Hi,

1. GTX 1080 will be supported when we update kernel to at least [version 4.12](https://kernelnewbies.org/Linux_4.12#Graphics) for Generic platform.

2.  Should work. Can you try to make a test with [nightly builds](http://nightly.builds.lakka.tv/) if you reproduce the issue.

3. It is using [GMA 3600](https://en.wikipedia.org/wiki/Intel_GMA#GMA_3600) which based on PowerVR and there is no accelerated driver for Linux. 

Thank you.

MontyGvMC, 2018-09-15:
Thank you for the information why one and 3 will not work.

I tested the nightly builds on the second one
I used the images 
a: Lakka-Generic.i386-2.2-devel-20180915112547-r28217-g74bf6c6.img.gz
b: Lakka-Generic.x86_64-2.2-devel-20180915112227-r28217-g74bf6c6.img.gz

a does not work, this time even worse, all I see is a black screen, not even the lakka logo
b does work great and perfectly

Do I have some missunderstanding about the difference between i386 and x86?
Should i386 work? I thought i386 just means 32 bit (like x86)

- Providing logs will be improved ... .-
Well, I am testing Lakka on old notebooks. Instead of live ssh and searching for the ip address (if network is available) I prefere the [modification of the commandline.txt/system.cfg to get the terminal](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/) (i have a monitor and keyboard) and write the logs to another usb stick...
Unfortunatly on a I didn't even manage to get into the terminal yet.

--- GMA 3600 accelerated drivers ---
Will start hearing around, may I can find a solution :-) I really want to get lakka/retroarch running on that hardware.

gouchi, 2018-09-15:
I tried Lakka-Generic-i386-2.1.1  and it is working correctly.

Can you try to boot with 
`live ssh` or `live textmode` to get some log at least [Lakka system and graphic card log](http://www.lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

MontyGvMC, 2018-09-15:
Unfortunatly I can't find a way too get the nightly generic i386 image to do anything.

I can't find an IP address, lakka.local isn't pingable,

no clue what live textmode should do,

adding tty and retroarch=0 to the commandline does not work

The screen keeps black after leaving the Boot: input Option (live vor installer).
Not even the lakka Berry logo is showing up.

And I recognized that the resizing of the live partition in first Boot was not there too.

MontyGvMC, 2018-09-15:
If you have any other suggestions what I could try I will do.

MontyGvMC, 2018-09-16:
I managed to use live ssh as well as live textmode with the x86_x64 IMG (to check if I am doing something wrong).

Doing the same with the i386 just gives me a black screen doing nothing.

If you are interested I can offer further experimenting up in recommendations what I should do next

or

if you could provide/suggest a running base image I could offer you SSH access to the maschine and you could give yourself a try.

gouchi, 2018-09-16:
`live textmode` should give you access to a terminal.

Try to boot then with `live textmode  debugging`, if still doesn't work you shoud try to use [break parameters](https://openelec.tv/documentation/configuration/boot-arguments).

Also is it possible to boot with another distro and get the output of `lspci -nnk | grep VGA ` ? So that we know exactly which graphic card you have.

Thank you.

MontyGvMC, 2018-09-16:
Grabbed from the Acer Aspire 1551 via ssh using the 2.2 devel x86_x64 generic image
I will keep trying to get the logs using the i386

`lspci -nkk | grep VGA`

`01:05.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] RS880M [Mobility Radeon HD 4225/4250] [1002:9712]
`
no clue why this is bold
--------------------

`lspci -nkk`

`00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] RS880 Host Bridge [1022:9601]
	Subsystem: Acer Incorporated [ALI] RS880 Host Bridge [1025:0429]
00:01.0 PCI bridge [0604]: Acer Incorporated [ALI] AMD RS780/RS880 PCI to PCI bridge (int gfx) [1025:9602]
00:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] RS780/RS880 PCI to PCI bridge (PCIE port 0) [1022:9604]
	Kernel driver in use: pcieport
00:05.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] RS780/RS880 PCI to PCI bridge (PCIE port 1) [1022:9605]
	Kernel driver in use: pcieport
00:11.0 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 SATA Controller [AHCI mode] [1002:4391]
	Subsystem: Acer Incorporated [ALI] SB7x0/SB8x0/SB9x0 SATA Controller [AHCI mode] [1025:0429]
	Kernel driver in use: ahci
00:12.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller [1002:4397]
	Subsystem: Acer Incorporated [ALI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller [1025:0429]
	Kernel driver in use: ohci-pci
00:12.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB EHCI Controller [1002:4396]
	Subsystem: Acer Incorporated [ALI] SB7x0/SB8x0/SB9x0 USB EHCI Controller [1025:0429]
	Kernel driver in use: ehci-pci
00:13.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller [1002:4397]
	Subsystem: Acer Incorporated [ALI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller [1025:0429]
	Kernel driver in use: ohci-pci
00:13.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB EHCI Controller [1002:4396]
	Subsystem: Acer Incorporated [ALI] SB7x0/SB8x0/SB9x0 USB EHCI Controller [1025:0429]
	Kernel driver in use: ehci-pci
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 SMBus Controller [1002:4385] (rev 41)
00:14.1 IDE interface [0101]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 IDE Controller [1002:439c] (rev 40)
	Subsystem: Acer Incorporated [ALI] SB7x0/SB8x0/SB9x0 IDE Controller [1025:0429]
	Kernel driver in use: pata_atiixp
00:14.2 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 Azalia (Intel HDA) [1002:4383] (rev 40)
	Subsystem: Acer Incorporated [ALI] SBx00 Azalia (Intel HDA) [1025:0429]
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 LPC host controller [1002:439d] (rev 40)
	Subsystem: Acer Incorporated [ALI] SB7x0/SB8x0/SB9x0 LPC host controller [1025:0429]
00:14.4 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] SBx00 PCI to PCI Bridge [1002:4384] (rev 40)
00:16.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller [1002:4397]
	Subsystem: Acer Incorporated [ALI] SB7x0/SB8x0/SB9x0 USB OHCI0 Controller [1025:0429]
	Kernel driver in use: ohci-pci
00:16.2 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD/ATI] SB7x0/SB8x0/SB9x0 USB EHCI Controller [1002:4396]
	Subsystem: Acer Incorporated [ALI] SB7x0/SB8x0/SB9x0 USB EHCI Controller [1025:0429]
	Kernel driver in use: ehci-pci
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 10h Processor HyperTransport Configuration [1022:1200]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 10h Processor Address Map [1022:1201]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 10h Processor DRAM Controller [1022:1202]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 10h Processor Miscellaneous Control [1022:1203]
	Kernel driver in use: k10temp
	Kernel modules: k10temp
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 10h Processor Link Control [1022:1204]
01:05.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] RS880M [Mobility Radeon HD 4225/4250] [1002:9712]
	Subsystem: Acer Incorporated [ALI] RS880M [Mobility Radeon HD 4225/4250] [1025:0429]
	Kernel driver in use: radeon
01:05.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] RS880 HDMI Audio [Radeon HD 4200 Series] [1002:970f]
	Subsystem: Acer Incorporated [ALI] RS880 HDMI Audio [Radeon HD 4200 Series] [1025:0429]
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel
03:00.0 Ethernet controller [0200]: Qualcomm Atheros AR8151 v1.0 Gigabit Ethernet [1969:1073] (rev c0)
	Subsystem: Acer Incorporated [ALI] AR8151 v1.0 Gigabit Ethernet [1025:040d]
	Kernel driver in use: atl1c
06:00.0 Network controller [0280]: Broadcom Limited BCM43225 802.11b/g/n [14e4:4357] (rev 01)
	Subsystem: Foxconn International, Inc. BCM43225 802.11b/g/n (T77H103.00 Wireless Half-size Mini PCIe Card) [105b:e021]
[... 819 lines cut ...]
[   46.151907] systemd[1]: Listening on Journal Socket.
[   46.151966] systemd[1]: Listening on udev Kernel Socket.
[   46.172932] systemd[1]: Created slice System Slice.
[   46.174436] systemd[1]: Starting Create list of required static device nodes for the current kernel...
[   46.227119] applespi: loading out-of-tree module taints kernel.
[... cut ...]
[   47.778082] Disabling lock debugging due to kernel taint
[   47.792767] input: HDA ATI SB Mic as /devices/pci0000:00/0000:00:14.2/sound/card0/input10
[   47.799440] input: HDA ATI SB Headphone as /devices/pci0000:00/0000:00:14.2/sound/card0/input11
[   47.938578] wlan0: Broadcom BCM4357 802.11 Hybrid Wireless Controller 6.30.223.271 (r587334)
[   47.938581] 
[   48.569423] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   48.659347] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)
`

MontyGvMC, 2018-09-16:
Finally the logs from the not working 2.2 devel generic i386 image

`lspci -nnk | grep VGA`

`01:05.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] RS880M [Mobility Radeon HD 4225/4250] [1002:9712]`

--------------

`dmesg`

`[    0.000000] Linux version 4.11.12-rt14 (vudiq@helsinki) (gcc version 7.3.0 (GCC) ) #1 SMP PREEMPT RT Fri Sep 7 00:15:42 CEST 2018
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009dc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000d0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000afe8ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000afe90000-0x00000000afeacfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000afead000-0x00000000afeaefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000afeaf000-0x00000000afefefff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000aff00000-0x00000000bfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fff00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000013fffffff] usable
[    0.000000] Notice: NX (Execute Disable) protection cannot be enabled: non-PAE kernel!
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: Acer            Aspire 1551                    /Aspire 1551                    , BIOS V1.11           05/14/2010
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0xafe90 max_arch_pfn = 0x100000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-CFFFF write-protect
[    0.000000]   D0000-DFFFF uncachable
[    0.000000]   E0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000000 mask FFFF80000000 write-back
[    0.000000]   1 base 000080000000 mask FFFFE0000000 write-back
[    0.000000]   2 base 0000A0000000 mask FFFFF0000000 write-back
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] TOM2: 0000000140000000 aka 5120M
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] e820: update [mem 0xb0000000-0xffffffff] usable ==> reserved
[    0.000000] found SMP MP-table at [mem 0x000f8080-0x000f808f] mapped at [c00f8080]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] initial memory mapped: [mem 0x00000000-0x023fffff]
[    0.000000] Base memory trampoline at [c0099000] 99000 size 16384
[    0.000000] BRK [0x02194000, 0x02194fff] PGTABLE
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000F8000 000024 (v02 PTLTD )
[    0.000000] ACPI: XSDT 0x00000000AFE9C61F 00007C (v01 ACRSYS ACRPRDCT 06040000 INNA 00000000)
[    0.000000] ACPI: FACP 0x00000000AFEAC0B5 0000F4 (v03 AMD    GUAM     06040000 AMD  000F4240)
[    0.000000] ACPI: DSDT 0x00000000AFE9C69B 00FA1A (v01 ATI    SB800    06040000 MSFT 03000001)
[    0.000000] ACPI: FACS 0x00000000AFEAEFC0 000040
[    0.000000] ACPI: FACS 0x00000000AFEAEFC0 000040
[    0.000000] ACPI: SLIC 0x00000000AFEAC21D 000176 (v01 ACRSYS ACRPRDCT 06040000 ANNI 00000001)
[    0.000000] ACPI: EINJ 0x00000000AFEAC393 0001B0 (v01 PTL    WHEAPTL  06040000 PTL  00000001)
[    0.000000] ACPI: HEST 0x00000000AFEAC543 0002E4 (v01 PTL    WHEAPTL  06040000 PTL  00000001)
[    0.000000] ACPI: BERT 0x00000000AFEAC827 000030 (v01 PTL    WHEAPTL  06040000 PTL  00000001)
[    0.000000] ACPI: SSDT 0x00000000AFEAC857 0000E1 (v01 wheaos wheaosc  06040000 INTL 20050624)
[    0.000000] ACPI: ERST 0x00000000AFEAC938 000270 (v01 PTL    WHEAPTL  06040000 PTL  00000001)
[    0.000000] ACPI: SSDT 0x00000000AFEACBA8 000386 (v01 AMD    POWERNOW 06040000 AMD  00000001)
[    0.000000] ACPI: APIC 0x00000000AFEACF2E 00005E (v01 PTLTD  ? APIC   06040000  LTP 00000000)
[... 694 lines cut ...]
    HDA ATI SB, ALC271X Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=SB,DEV=0
    HDA ATI SB, ALC271X Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=SB,DEV=0
    HDA ATI SB, ALC271X Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=SB,DEV=0
    HDA ATI SB, ALC271X Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=SB,DEV=0
    HDA ATI SB, ALC271X Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=SB,DEV=0
    HDA ATI SB, ALC271X Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=SB,DEV=0
    HDA ATI SB, ALC271X Digital
    IEC958 (S/PDIF) Digital Audio Output
hdmi:CARD=HDMI,DEV=0
    HDA ATI HDMI, HDMI 0
    HDMI Audio Output`

------------

gouchi, 2018-09-17:
Your card is supported by the [radeon](https://cateee.net/lkddb/web-lkddb/DRM_RADEON.html) kernel module.

Is it possible to [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka/#lakka-system-logs) also ?

Thank you.

MontyGvMC, 2018-09-17:
Sure, later that day.

But which one actually would be the Lakka system log? Thought dmesg (as above) would be lakka system log.

gouchi, 2018-09-17:
What is written for [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka/#lakka-system-logs) ;-)

`systemctl stop retroarch.service ; retroarch -v`

or

`systemctl stop retroarch.service ; LIBGL_DEBUG=verbose retroarch --menu --verbose >> log.txt 2>&1`

MontyGvMC, 2018-09-17:
> What is written for [Lakka system log](http://www.lakka.tv/doc/Troubleshooting-Lakka/#lakka-system-logs) ;-)

Well, what should I say... you wouldn't believe that people sometimes think I appear confused...

retroarch/lakka system logs from the i386 image

`[INFO] RetroArch 1.7.4 (Git 1155e11)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 
Built: Sep  9 2018
[INFO] Version: 1.7.4
[INFO] Git: 1155e11
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
function is no-op
function is no-op
function is no-op
function is no-op
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault (core dumped)`

gouchi, 2018-09-30:
You are still getting segmentation fault with i386 build.

The only difference I see between i386 and x86_64 builds regarding radeon/amd gpu is that we enabled [CONFIG_DRM_AMDGPU_SI](https://cateee.net/lkddb/web-lkddb/DRM_AMDGPU_SI.html) and [CONFIG_DRM_AMDGPU_CIK](https://cateee.net/lkddb/web-lkddb/DRM_AMDGPU_CIK.html) on x86_64 but it is not related to your card.

If the log is not too long post here otherwise on pastebin services.

We you can try to get more information about drm if you boot with
`live ssh drm.debug=0x1e log_buf_len=1M`

then in the terminal
`journalctl | grep drm`

But The log will be long...

MontyGvMC, 2018-10-02:
Kind a busy currently. I provide the logs later that week...

MontyGvMC, 2018-10-07:
`journalctl | grep drm` -> not that long

`Oct 07 11:42:38 Lakka kernel: [drm] radeon kernel modesetting enabled.
Oct 07 11:42:38 Lakka kernel: [drm] initializing kernel modesetting (RS880 0x1002:0x9712 0x1025:0x0429 0x00).
Oct 07 11:42:38 Lakka kernel: [drm] register mmio base: 0xD0100000
Oct 07 11:42:38 Lakka kernel: [drm] register mmio size: 65536
Oct 07 11:42:38 Lakka kernel: [drm] Detected VRAM RAM=336M, BAR=256M
Oct 07 11:42:38 Lakka kernel: [drm] RAM width 32bits DDR
Oct 07 11:42:38 Lakka kernel: [drm] radeon: 336M of VRAM memory ready
Oct 07 11:42:38 Lakka kernel: [drm] radeon: 512M of GTT memory ready.
Oct 07 11:42:38 Lakka kernel: [drm] Loading RS780 Microcode
Oct 07 11:42:38 Lakka kernel: [drm:r600_init] *ERROR* Failed to load firmware!
Oct 07 11:42:38 Lakka kernel: [drm] radeon: finishing device.
Oct 07 11:42:38 Lakka kernel: [drm] radeon: ttm finalized
Oct 07 11:42:38 Lakka kernel: [drm] amdgpu kernel modesetting enabled.
Oct 07 11:45:57 Lakka kernel: Kernel command line: root=/dev/ram0 rdinit=/init usbcore.autosuspend=-1 BOOT_IMAGE=/KERNEL boot=UUID=1509-2754 disk=UUID=96c20583-90a5-441f-bce0-84dc9b87dd9a portable quiet vt.global_cursor_default=0 loglevel=2 ssh textmode debugging nosplash drm.debug=0x1e log_buf_len=1M
Oct 07 11:45:57 Lakka kernel: [drm] radeon kernel modesetting enabled.
Oct 07 11:45:57 Lakka kernel: [drm] initializing kernel modesetting (RS880 0x1002:0x9712 0x1025:0x0429 0x00).
Oct 07 11:45:57 Lakka kernel: [drm] register mmio base: 0xD0100000
Oct 07 11:45:57 Lakka kernel: [drm] register mmio size: 65536
Oct 07 11:45:57 Lakka kernel: [drm] Detected VRAM RAM=336M, BAR=256M
Oct 07 11:45:57 Lakka kernel: [drm] RAM width 32bits DDR
Oct 07 11:45:57 Lakka kernel: [drm] radeon: 336M of VRAM memory ready
Oct 07 11:45:57 Lakka kernel: [drm] radeon: 512M of GTT memory ready.
Oct 07 11:45:57 Lakka kernel: [drm] Loading RS780 Microcode
Oct 07 11:45:57 Lakka kernel: [drm:r600_init] *ERROR* Failed to load firmware!
Oct 07 11:45:57 Lakka kernel: [drm] radeon: finishing device.
Oct 07 11:45:57 Lakka kernel: [drm] radeon: ttm finalized
Oct 07 11:45:57 Lakka kernel: [drm] amdgpu kernel modesetting enabled.
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] Loading Nouveau with parameters:
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... tv_disable   : 0
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... ignorelid    : 0
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... duallink     : 1
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... nofbaccel    : 0
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... config       : (null)
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... debug        : (null)
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... noaccel      : 0
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... modeset      : -1
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... runpm        : -1
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... vram_pushbuf : 0
Oct 07 11:45:57 Lakka kernel: [drm:nouveau_drm_init] ... hdmimhz      : 0`

gouchi, 2018-10-07:
I did not see this error before
```
[    1.693980] radeon 0000:01:05.0: Direct firmware load for radeon/R600_rlc.bin failed with error -2
[    1.693982] r600_cp: Failed to load firmware "radeon/R600_rlc.bin"
[    1.693990] [drm:r600_init] *ERROR* Failed to load firmware!

```

It is working on x86_64 because [kernel firmwares are copied only for x86_64 not i386](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/linux/package.mk#L114).

ToKe79, 2019-06-27:
@MontyGvMC can you try [this 32bit image](http://nightly.builds.lakka.tv/special_builds/Generic.i386_add-kernel-firmware/).
Changes: https://github.com/ToKe79/Lakka-LibreELEC/commit/6257b04678bfb1b3491a85dda679eb4bfee05a67

gouchi, 2019-06-27:
@ToKe79 We should add `CONFIG_EXTRA_FIRMWARE_DIR="external-firmware"` to i386 kernel config.

ToKe79, 2019-06-27:
I updated the image with correct linux.config.

MontyGvMC, 2019-06-28:
I assume on wednesday evening I can give it a try

ToKe79, 2019-07-10:
@MontyGvMC any luck with the testing?

MontyGvMC, 2019-07-11:
Unfortunatly not yet and will have to to wait until next week.

MontyGvMC, 2019-07-20:
Did download the linked image and started it from usb in live mode.

Same thing: not even reaching the lakka berry logo on boot.

Logs will take some more time. Again having troubles to get into the tty or ssh.

ToKe79, 2019-07-22:
which setup did you try the linked image on? If the one with GTX 1080, try [this image](http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/Generic.i386/). However there is issue with some systemd units, which fail on the 3rd boot (1st boot - /storage expansion - OK, 2nd boot - OK, 3rd boot - failing services). I am investigating this.

MontyGvMC, 2019-07-22:
it was the Acer Aspire 1551

gouchi, 2019-07-23:
May you try to provide some [log](https://github.com/libretro/Lakka-LibreELEC/issues/543#issuecomment-422082673) by booting with `live ssh textmode` in the boot prompt.

Thank you.

MontyGvMC, 2019-07-24:
> Logs will take some more time. Again having trouble to get into the tty or ssh.

Like mentioned and like in September 2018 having troubles to access the logs....

Either I am too dumb to enter "live textmode ssh" correctly or it does not work as expected (at least by me).

gouchi, 2019-07-24:
Just burn the image and in the [boot prompt](https://youtu.be/8rMvf3tfjFk?t=121) enter `live ssh textmode
`

closed 2020-08-18 by Ntemis