libretro/Lakka-LibreELEC #547: Unable to boot - Stuck on flower screen

Frankenburger, 2018-10-02:
I just grabbed an Intel NUC with a J5005 Pentium Silver CPU. I’ve tried both the latest official 2.1 release and the latest nightly builds of Lakka. Both builds work fine on my old Celeron NUC, but neither boot into Retroarch with the new NUC. Log below

Lakka (official): devel-20181002000707-r28220-g34ec56d (Generic.x86_64) Lakka:~ # systemctl stop retroarch.service ; retroarch -v
[INFO] RetroArch 1.7.4 (Git 1155e11)
[INFO] === Build ======================================= Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AES Built: Sep 9 2018 [INFO] Version: 1.7.4
[INFO] Git: 1155e11
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565. [INFO] Redirecting save file to “/storage/savefiles/.srm”.
[INFO] Redirecting savestate to “/storage/savestates/.state”.
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver …
[ERROR] [KMS]: Couldn’t find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
[INFO] [GL]: Vendor: (null), Renderer: (null). [INFO]
[GL]: Version: (null). Segmentation fault (core dumped)

ToKe79, 2018-10-08:
Did you try both HDMI ports?

gouchi, 2018-10-10:
Also what is possible to get [graphic card logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/#graphic-card-logs) ?

Thank you.

closed 2019-06-26 by ToKe79