libretro/Lakka-LibreELEC #549: AMD Ryzon 2400G (Vega 11 APU)

sheepchen, 2018-10-04:
- Which version of Lakka are you using?
Lakka-Generic.x86_64-2.1.1 (Linux)

- What system hardware are you using?
Gigabyte AB350N-Gaming WiFi
AMD Ryzen 2400G with 2x8GB RAM
GPU: APU Radeon RX Vega 11

- What did you do?
(- What did you expect to happen?)
- What happened instead?
Live-Boot. After resizing the stick and restarting the system hangs at the Lakka-Logo for half an hour. After this the logo disapiers and the is only a black screen (but there is a signal). The system is reading from the stick the whole time. Let it sit for 3 hours with no result.

ToKe79, 2018-10-04:
Hi,
your GPU is not supported by Lakka currently. Lakka is aimed primary for low spec computers (http://www.lakka.tv/lakka-is-low-cost/). In your case I would suggest installing the RetroArch "suite" (RetroArch + libretro cores) in the OS of your choice (Linux / Windows / ...).

gouchi, 2019-03-08:
@sheepchen can you try this [testing image](https://drive.google.com/drive/folders/128MaH09bjVcB2pR4Kj2XxdBabR-oDP75) ?

Thank you.

closed 2019-07-02 by ToKe79