libretro/Lakka-LibreELEC #554: Lakka-Generic_x86_x64 network issues

MontyGvMC, 2018-10-10:
- Which version of Lakka are you using?   
Lakka_Generic_x86_x64_2.1.1.img

- What system hardware are you using? 
Acer Aspire 1551 with WLAN: Atheros AR5B93 and LAN: Atheros AR8151

- What did you do?
I installed the image, entered WiFi credentials. I downloaded thumbnails (so WiFi did work).

I wanted to place some ROMS via SSH. 
LAKKA host was not visbile nor reachable via SSH over WiFi (though it was connected since I was able to download the thumbnails and WiFi settings showed the online flag).
I changed from WiFi to LAN. LAKKA was visible and reachable via SSH over LAN.

When entering the Settings > WiFi menu in RetroArch a scan starts.
Sometimes the result shows the expected WLAN networks, sometimes it only shows one entry with some cryptic signs and WiFi does not work.

Which logs are of interest?

ToKe79, 2018-10-10:
When you were using WLAN, did you try to connect using the hostname (e.g. `ssh root@lakka`) or the IP address (e.g. `ssh root@192.168.0.111`)? Sometimes the wifi router might not resolve the hostname to IP address, whatever the reason is.

MontyGvMC, 2018-10-10:
I am always trying the hostname (lakka) since my home router (Huawei B529s) seems to have issues showing ip adresses of connected devices.

To figure out the ip of my lakka laptop I use the [free ip scanner](http://www.eusing.com/ipscan/free_ip_scanner.htm). When connected via LAN Lakka shows up in the scan results. When connected via WiFi it doesnt so I am even unable to figure out the ip adress.

ToKe79, 2018-10-10:
The IP address of Lakka device should show up in some status/information menu in the RetroArch XMB frontend (sorry - do not know by memory which one it is).

ToKe79, 2018-10-10:
Found it in here: https://github.com/libretro/Lakka-LibreELEC/wiki/Finding-the-IP-of-your-Lakka-box

> You can have access to your IP Address via the menu **Information** > **Network Information**.

MontyGvMC, 2018-10-15:
I managed to try it out.
Thanks for the hint that ip adress is shown. Yes, via WLAN SSH works using directly the IP address.

The interesting point I wanted to reproduce but failed was that sometimes even the free ip scanner nor the ip scanner I used were unable to show my lakka network (though they were connected and I could download something from within retroarch). But they showed some interesting unknown devices in my personal network. But those are other stories...

One thing I recognized is that it seems my Lakka laptop seems to use an all zero mac-address.

kivutar, 2018-12-25:
It also happens to me on some home networks only

closed 2018-12-25 by kivutar