libretro/Lakka-LibreELEC #556: Update to LibreELEC v8.90.006

RobLoach, 2018-10-10:
https://libreelec.tv/2018/10/libreelec-leia-v8-90-006-alpha/

https://github.com/LibreELEC/LibreELEC.tv/tree/8.90.006

natinusala, 2018-10-10:
FYI the Rockchip `BOARD` variable is similar to LE's `UBOOT_SYSTEM` - the first should be replaced by the latter when updating (I can do it if needed).

kivutar, 2018-10-17:
We should do this in a separate Lakka 3.0 branch. And I suggest that we start from a clean LE 8.9 and port over everything again.
Starting from easy stuff like x86_64 and RPi2.

gpapp, 2018-10-31:
> We should do this in a separate Lakka 3.0 branch. And I suggest that we start from a clean LE 8.9 and port over everything again.
> Starting from easy stuff like x86_64 and RPi2.

I created a fork and started from the latest LibreELEC v8.9 here. https://github.com/gpapp/Lakka-LibreELEC/tree/Lakka-V3.0

bpereto, 2018-12-23:
I started from a clean LibreElec v8.9here:
https://github.com/bpereto/Lakka/tree/lakka-3.0

And started porting all over to get it work under an new intel nuc i3 coffee lake.
It is currently building for Generic and will keep you updated.

Changes I made:
* Added packages:
  * `lakka`
  * `libretro/*`
  * sixpair
  * empty
  * joyutils
* Added `PKG_URL` to all libretro packages (`PKG_GIT_URL` + `archive/PKG_VERSION`)
* Re-added `strip_lto` and `strip_gold` functions to `config/functions`

Ntemis, 2018-12-23:
Good work @bpereto when you manage to build any other projects throw them our way so we can test.
I liked what i saw thus far

bpereto, 2018-12-24:
A christmas present :gift: for all and @Ntemis :)
* Working Image
* System boots (besides I have to press ctrl + F3 to show retroarch instead of a blank screen)
* Cores are adjusted and are building, except:
  * `citra` (some cmake errors)
  * `dosbox-svn` git/pkg url not found/unauthorized (lakka mirror)

happy testing. :tada: 
https://data.sandchaschte.ch/pub/lakka/Lakka-Generic.x86_64-3.0-devel-20181224102324-8f8eaf0-sandchaschte.img.gz

natinusala, 2018-12-24:
Nice job !

The dosbox-svn repository is https://github.com/fr500/dosbox-svn right ? It works for me

bpereto, 2018-12-24:
right, I messed up the PKG_URL :) - I should controll all PKG_URLs.

My testing for today:
* sound does not work (Alsa /dev/snd/seq not found) - sequencer added with https://github.com/bpereto/Lakka/commit/c45ab5c384eb891542b513fe286888faaa5ee68b but needs a new build
  I don't know how to to test this :/ - HDMI output with an ubuntu live image does work.
* Blank screen on boot: xorg and retroarch have concurrency issues. display works fine when disabling xorg (`systemctl stop xorg; systemctl disable xorg; systemctl restart retroarch`

natinusala, 2018-12-24:
Why would you use xorg though ? Just use KMS with retroarch

bpereto, 2018-12-25:
@natinusala 
when i set `DISPLAYSERVER="no"`, then there are missing libs like libXrandr or mesa (libGL.so) when starting retroarch:
```
Lakka:~ # retroarch -v
retroarch: error while loading shared libraries: libGL.so.1: cannot open shared object file: No such file or directory
```

Where should I define the dependencies?

natinusala, 2018-12-25:
Missing libXrandr is normal, however you shouldn't need it so you will need to why is Retroarch linked against it and solve that here.

As for mesa, it should be in the same file as `DISPLAYSERVER` : `OPENGL="mesa"` and `OPENGLES`="no"`

bpereto, 2018-12-25:
https://github.com/bpereto/Lakka/blob/lakka-3.0/projects/Generic/options#L66
``` 
 # OpenGL(X) implementation to use (no / mesa)
    OPENGL="mesa"

  # OpenGL-ES implementation to use (no / bcm2835-driver / gpu-viv-bin-mx6q)
    OPENGLES="no"

  # Displayserver to use (x11 / no)
    DISPLAYSERVER="no"
```
probably I must do a clean build for 4 hours :/

bpereto, 2018-12-26:
Done that, runned into issues with mesa.
Seems to be that the updated mesa package now use the meson buildsystem. theres a bunch of tweaking nessecary.

* added a patch to make mesa build (https://github.com/bpereto/Lakka/blob/7a75177633bb2be989cd6fe52fa8babec0fb79b5/packages/graphics/mesa/patches/001-missing-include-egldevice.patch)
* adjusted PKG_DEPENDS like already in place in Lakka 2.2
* Added some missing protos for those settings

Result:
* Working Image, boot from kms into retroarch :)
* no sound :( - audio is not muted, audio device switching changes nothing..
```
Lakka:~/roms/nas/n64 # retroarch -v
[INFO] RetroArch 1.7.5 (Git fae431a70b)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AES 
Built: Dec 26 2018
[INFO] Version: 1.7.5
[INFO] Git: fae431a70b
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 3 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 48 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 3840x2160.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x7fb9a012ca10.
[INFO] [KMS]: New FB: 3840x2160 (stride: 15360).
```

bpereto, 2018-12-27:
Latest status update
https://github.com/bpereto/Lakka/tree/lakka-3.0

Work so far:
* linux 4.19
* KMS / Graphic
* Sound over HDMI
* All cores building

Bonus:
* Added Vulkan API https://github.com/bpereto/Lakka/commit/760c5796f0294f7150a585debe1aab0306ba73a6

the current state works for me, so you can do your testing and pull the branch.

In the next days I can finally do some additional testing (some gaming :smile: )
I can provide a Image if someone wants it.

natinusala, 2018-12-27:
Vulkan support is cool, but be aware that it won't work with every graphics card.

Actually I know that mesa doesn't have any support for Vulkan on NVIDIA cards (yet), and I'm not sure about AMD. You might want to at least try those two before releasing Vulkan on Lakka.

5schatten, 2018-12-28:
@natinusala @bpereto 
Vulkan *should* work fine if you use Mesa for Intel & AMD GPUs. The latter maybe needs some tweaks like force amdgpu as kernel driver. Nvidia & Vulkan works also fine then you'll need the vendor drivers & probably X11.

jasond815, 2019-01-14:
> Latest status update
> https://github.com/bpereto/Lakka/tree/lakka-3.0
> 
> Work so far:
> 
> * linux 4.19
> * KMS / Graphic
> * Sound over HDMI
> * All cores building
> 
> Bonus:
> 
> * Added Vulkan API [bpereto/Lakka@760c579](https://github.com/bpereto/Lakka/commit/760c5796f0294f7150a585debe1aab0306ba73a6)
> 
> the current state works for me, so you can do your testing and pull the branch.
> 
> In the next days I can finally do some additional testing (some gaming 😄 )
> I can provide a Image if someone wants it.

bpereto I would love your latest image if you could.

cowboysmall, 2019-01-19:
> In the next days I can finally do some additional testing (some gaming  )
How did the testing go?

> I can provide a Image if someone wants it.
I'd be very interested in the latest image too.

bpereto, 2019-01-20:
**Testing:**   
Seems to be working, altrough my i3 has some issues with the ppsspp core, but this seems to be core related, or unoptimized for x86.
* SNES works wonderful for me
* My Xbox Controller 360 worked instantly with the wireless adapter from microsoft.
* At the beginning of my play-around the SNES it lagged/blacked out sometimes, but this seems to be related to the temperature of my box and cpu frequency drops.
* The Controller Key mapping is a little weird, I had to change that.

Not tested:
* Network Play

My latest Image can be found here:
https://drive.google.com/open?id=128MaH09bjVcB2pR4Kj2XxdBabR-oDP75

RobLoach, 2019-02-02:
[LibreELEC 9.0.0 is out](https://libreelec.tv/2019/02/libreelec-leia-9-0-0-release/) :wink:

cowboysmall, 2019-02-08:
I want to create my own Lakka-LibreELEC project by taking the following steps:

- fork the latest LibreELEC release (9.0.0)
- add all the Lakka-LibreELEC packages 
- add any other packages I want

My reason for doing this is that I want to be able to pull / merge updates from LibreELEC easily. Also I want to be able to create an up-to-date Lakka image easily, with any custom changes I want (package upgrades, package additions, etc.). 

Can someone point me at documentation that would show me how to create Lakka-LibreELEC from a clean LibreELEC?

Thanks in advance,

J.K.

RobLoach, 2019-02-08:
* fork the latest LibreELEC release (9.0.0)
* add all the Lakka-LibreELEC packages
* add any other packages I want

;-)

5schatten, 2019-02-08:
@cowboysmall 
well understand how LE hanldes [packages](https://github.com/LibreELEC/LibreELEC.tv/tree/master/packages) and work your way through. LibreELEC is basically LFS based so if you want create a custom build just add your stuff, satisfy the dependencies, configure and optimize it.

So unless you don't have a lot of time to spent wait for Lakka 3.0 the fork it and update this build since it's more suitable for you than altering LE to your own Lakka custom build.

cowboysmall, 2019-02-08:
@5schatten 

thanks for the response. I'll take the approach you suggest. As a matter of interest, any idea when Lakka 3.0 will be released?

J.K.

5schatten, 2019-02-09:
@cowboysmall 
I guess when it's done... I'm not part of the dev team but since the underlaying build system and else has been overhauled you have to adjust pretty much every package to rebase Lakka.

marcfehling, 2019-04-23:
[LibreELEC 9.0.1 is out](https://libreelec.tv/2019/02/libreelec-leia-9-0-1-mr/)

marcfehling, 2019-05-22:
[LibreELEC 9.0.2 is out](https://libreelec.tv/2019/05/libreelec-leia-9-0-2-mr/)

ToKe79, 2019-06-19:
I started working on Lakka based on the libreelec-9.0 branch (https://github.com/libretro/Lakka-LibreELEC/tree/Lakka-LE9.0). Testing images: http://nightly.builds.lakka.tv/Lakka-LE9.0-experimental/

Ntemis, 2019-06-21:
What socs are supported thus far?

ToKe79, 2019-06-21:
So far I was able to build (but not test):
- Generic.x86_64
- RPi2.arm
- S905.arm

You are free to checkout the branch and adjust where necessary and commit your changes (I would prefer PRs, but I do not think the branch is protected from direct pushes).

I have a quite positive feedback on the RPi2 (minor adjustments were necessary and already implemented in 2d710d9411d98a90400be53ca985abcab6fa92d6).

I am storing the finished images at http://nightly.builds.lakka.tv/Lakka-LE9.0-experimental/

natinusala, 2019-06-21:
I have two requests for Lakka 3.0:
- aarch64 targets must build (and work)
- xorg should not be included on any platform that we currently support _but_ it should be working if we set the `DISPLAYSERVER` variable (this is to make the Jetson / Switch port easier in the future)

Hiki1, 2019-07-11:
I've tested the image _Lakka-RPi2.arm-2.9-devel-20190710170355-3f7ff53_ @ToKe79 is working on, and I have no video over composite (works over HDMI). It may be connected to this issue: https://github.com/raspberrypi/firmware/issues/1166
I'm using Raspberry Pi 3 Model B+

closed 2020-08-18 by Ntemis