libretro/Lakka-LibreELEC #557: Building different versions of Lakka using docker fail while unpacking sameboy.

grillo-del-mal, 2018-10-11:
#### Which version of Lakka are you using?
* I tested this with the master branch, and the latest 2 tags (2.1 and 2.1.1) 

#### What system hardware are you using?
* Tested this on a couple of x86 computers running Linux and Docker

#### What did you do?
* Tried to build lakka using the instructions on the dockerfile
```
docker build -t lakka .
docker run -it -v $(pwd):/root lakka
```
* Also tested with some enviorment variables
```
docker build -t lakka .
docker run -it -e DISTRO="Lakka" -e PROJECT="RPi" -e ARCH="arm" -v $(pwd):/root lakka
```

#### What did you expect to happen?
* Build process ends normally, I get my binaries ready to install.

#### What happened instead?
* Build process stops when unpacking the sameboy package
```
  UNPACK   sameboy
Cloning into '/root/build.Lakka-RPi.arm-2.2-devel/sameboy-179d3f6'...
remote: Enumerating objects: 4137, done.
remote: Total 4137 (delta 0), reused 0 (delta 0), pack-reused 4137
Receiving objects: 100% (4137/4137), 5.57 MiB | 514.00 KiB/s, done.
Resolving deltas: 100% (2970/2970), done.
Checking connectivity... done.
error: pathspec '179d3f6' did not match any file(s) known to git.
Makefile:27: recipe for target 'image' failed
make: *** [image] Error 1
```

#### Other tests
* Tried to run the process manually using the sameboy repository (guiding myself from what the `script/unpack` script does
  * Got the same error
```
$ clone https://github.com/libretro/SameBoy.git
Cloning into 'SameBoy'...
remote: Enumerating objects: 4137, done.
remote: Total 4137 (delta 0), reused 0 (delta 0), pack-reused 4137
Receiving objects: 100% (4137/4137), 5.57 MiB | 44.00 KiB/s, done.
Resolving deltas: 100% (2970/2970), done.
$ git checkout 179d3f6
fatal: not a git repository (or any parent up to mount point /)
$ git checkout 9ae8304
fatal: not a git repository (or any parent up to mount point /)
$ git checkout 8d69156
fatal: not a git repository (or any parent up to mount point /)
```
* It actually works if you use the head of the buildbot branch
* Also noticed that when searching the commit on the libretro/sameboy repo's github page, it always points to a branchless commit.

#### Posible fixes
* Recover the affected commits on the sameboy repository into new branches (only if someone have them locally)
  * https://help.github.com/articles/commit-exists-on-github-but-not-in-my-local-clone/#the-branch-that-contained-the-commit-was-deleted
* Change the package.mk to download the archived commit directly from github instead of using the git repository (like how it's done with the retroarch package)
  * https://github.com/libretro/Lakka-LibreELEC/blob/v2.1/packages/libretro/retroarch/package.mk
  * https://github.com/libretro/SameBoy/archive/179d3f6.tar.gz
  * https://github.com/libretro/SameBoy/archive/9ae8304.tar.gz
  * https://github.com/libretro/SameBoy/archive/8d69156.tar.gz

natinusala, 2018-10-11:
I think we should migrate all git based packages (which don't have submodules) to the targz method. This way we eliminate any issue when rebasing repositories, and we gain time by not cloning the repo each time the package is changed / building a clean image.

RobLoach, 2018-10-12:
> I think we should migrate all git based packages (which don't have submodules) to the targz method.

The issue is unrelated to Docker. Using the git hashes allows us to target a version that we know is working. In this case, we just need to update the hash. I'll make a PR.

RobLoach, 2018-10-12:
Sent a bunch of PRs up to Sameboy: https://github.com/LIJI32/SameBoy/pulls

RobLoach, 2018-10-12:
It's over at https://github.com/libretro/Lakka-LibreELEC/pull/559 ... The problem is the dependencies behind SameBoy. In order to get around it, we've made a `buildbot` branch with the dependencies directly in.

natinusala, 2018-10-12:
@RobLoach the targz method still uses the git hash but instead of doing a git clone + git checkout, it downloads the targz archive of that particular commit from GitHub. We just need to shift to the long hash and set `PKG_SOURCE_DIR` properly : 

```
PKG_NAME="retroarch"
PKG_VERSION="d628e01247ca9c908a7c5376bd1e1f8ad2a7cba2"
PKG_SITE="https://github.com/libretro/RetroArch"
PKG_URL="$PKG_SITE/archive/$PKG_VERSION.tar.gz"
PKG_SOURCE_DIR="RetroArch-$PKG_VERSION"
```

By the way the dependencies for sameboy are an issue for the buildbot but are not for us, I already have a package ready for rgbasm for instance

RobLoach, 2018-10-12:
Nice! Started making that one, and then gave up :wink:

natinusala, 2018-10-12:
Here it is : https://github.com/lakka-switch/Lakka-LibreELEC/blob/Lakka-V2.1-dev/packages/tools/rgbds/package.mk

RobLoach, 2018-10-12:
Last time I tested compiling rgbds, it complained about not having `flex`. Should we include that in the `PKG_DEPENDS_HOST`?

natinusala, 2018-10-12:
Sure, `flex:host` should do the trick

RobLoach, 2018-10-16:
Just pushed up a bunch of fixes to Sameboy upstream. Builds in GCC correctly now.

ToKe79, 2019-07-02:
fixed with #560

closed 2019-07-02 by ToKe79