libretro/Lakka-LibreELEC #564: (another) Lakka stuck in flower screen while booting

Isgolo, 2018-11-06:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 
2.1.1 32-bit
- What system hardware are you using?
ASRock P4i65G Intel 865G Chipset
Intel Pentium 4 3GHz
1Gb DDR400 RAM
ATI Radeon 9200SE 128Mb
SATA Seagate Barracuda 7200.7 HDD 80Gb
- What did you do?
Install and boot
- What did you expect to happen?
Boot..?
- What happened instead?
Stuck in Flower screen

Logs:
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/2554068/log.txt)
[graphiclog.txt](https://github.com/libretro/Lakka-LibreELEC/files/2554063/graphiclog.txt)
[dmesglog.txt](https://github.com/libretro/Lakka-LibreELEC/files/2554065/dmesglog.txt)
[audiolog.txt](https://github.com/libretro/Lakka-LibreELEC/files/2554077/audiolog.txt)

I've already seen a coulple posts with this problem, but the fixes for those post didnt work for me. 

If you need another log tell me how to get it, it took me ages to get this ones (never used linux before and i dont understand how to use the terminal, im just copying commands from the posts).

Im going to try now all nightly builds i can, just in case it works and ill report if some work.

Thanks in advance!

gouchi, 2018-11-10:
Can you try to [edit in the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface) and change to `video_monitor_index = "1"` ?

If it still doesn't work change also to `menu_shader_pipeline = "0"`.

Thank you.

Isgolo, 2018-11-10:
With Shader = 1
[log1.txt](https://github.com/libretro/Lakka-LibreELEC/files/2568519/log1.txt)

With shader = 0
[log2.txt](https://github.com/libretro/Lakka-LibreELEC/files/2568520/log2.txt)

I've also tried a couple more settings i found in other posts, and the error is always the same:
`RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [ERROR] :: Failed to compile vertex shader #0
RetroArch [ERROR] :: Failed to link program #0.
RetroArch [ERROR] :: GLSL stock programs failed to compile.
RetroArch [ERROR] :: [GL]: Failed to initialize shader, falling back to stock.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [ERROR] :: Failed to compile vertex shader #0
RetroArch [ERROR] :: Failed to link program #0.
RetroArch [ERROR] :: GLSL stock programs failed to compile.
RetroArch [ERROR] :: [GL]: Shader initialization failed.
Segmentation fault (core dumped)`

I dont know a lot about programming, but maybe the graphics card is not compatible with the version of OpenGL used?

gouchi, 2018-11-10:
You even get a segfault running with `video_monitor_index = "1"` only ? Is it possible to get a log with it ?

Lastly try  `video_monitor_index = "1"`  and `menu_driver = "rgui"`. If it does not work with this configuration, your graphic card is maybe too old to run RA.

Isgolo, 2018-11-10:
> You even get a segfault running with `video_monitor_index = "1"` only ? Is it possible to get a log with it ?
>
Yeah, both logs are with it. This one is with 'video_monitor_index =  "1"', 'menu_shader_pipeline = "0"' and 'menu_driver = "rgui"'
[log3.txt](https://github.com/libretro/Lakka-LibreELEC/files/2568733/log3.txt)
Here is my [retroarch.txt](https://github.com/libretro/Lakka-LibreELEC/files/2568737/retroarch.txt)
>If it does not work with this configuration, your graphic card is maybe too old to run RA.
>
Probably, it is a Radeon 9200SE 128Mb, I think the lastest version of OpenGL supported is 1.4. Is there an old version of retroarch with that version?

gouchi, 2018-11-11:
Yes you will need at least [opengl 2.0 support](https://github.com/libretro/RetroArch/issues/5188#issuecomment-316693720).

Moreover, we have disabled SDL so you can't use SDL video driver.  

I am afraid the only solution is to change your graphic card.

Isgolo, 2018-11-11:
All right, first ill try with integrated graphics of a couple motherboards i have laying around.
Also, if there is an older version of lakka using SDL drivers I'd like to try it, and I suggest leaving a link in the dowload page for anyone using REALLY old computers.
If there is not an older version and the integrated graphics do not work ill try to buy a card with openGL 2.0 (maybe the radeon 9500?)
Thanks for the help!

gouchi, 2018-11-11:
No problem you are welcome.

I am not sure we enabled SDL. Anyway using SDL is the last change as the performance won't be optimal.

You can try [Lakka 2.0](http://le.builds.lakka.tv/Generic.i386/) but  I think you will get the same result.

Yes it seems [radeon 9500 supports openGL 2.0](https://en.wikipedia.org/wiki/Radeon_9000_Series).

I will close this issue as your GPU is too old for RA.

closed 2018-11-11 by gouchi