libretro/Lakka-LibreELEC #570: Lakka meets segmentation fault when loading DOSbox games on 32bits PC

xuqingkuang, 2018-12-02:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?

2.1.1 for Generic PC 32bits

- What system hardware are you using?

Lenovo Q110 - Intel ATOM + Nvidia ION

- What did you do?

Loading the Theme Hospital game extract the package from https://gamesnostalgia.com/en/game/theme-hospital.

BTW: I tried another DOS game but all the same. 

- What did you expect to happen?

Game load normally.

- What happened instead?

RetroArch restart, and see segmentation fault in retroarch log.

**Informations attached**
RetroArch log: [retroarch-log.txt](https://github.com/libretro/Lakka-LibreELEC/files/2636470/retroarch-log.txt)
DOS Playlist: [DOS.lpl.txt](https://github.com/libretro/Lakka-LibreELEC/files/2636471/DOS.lpl.txt)
Theme Hospital Config: [dosbox.conf.txt](https://github.com/libretro/Lakka-LibreELEC/files/2636473/dosbox.conf.txt)

xuqingkuang, 2018-12-03:
Sorry the project seems is about the linux distribution, I create another issue in RetroArch project for track it - https://github.com/libretro/RetroArch/issues/7692

closed 2018-12-03 by xuqingkuang

gouchi, 2018-12-04:
@xuqingkuang May you try to see if you reproduce the issue with [nightly builds](http://nightly.builds.lakka.tv/) ?