libretro/Lakka-LibreELEC #571: Q) How to install the PC version on the hard drive?

zezba9000, 2018-12-03:
I can't find this anywhere. Everything I search for just pulls up Pi stuff.
I hope I don't have to use clonezilla ?

RobLoach, 2018-12-03:
Lakka is an operating system that runs [RetroArch](https://retroarch.com/). If you're looking to use this on your personal computer, visit https://retroarch.com/ and download the application there.

zezba9000, 2018-12-03:
No I want to install Lakka on my primary disk. Not run it off a USB. My computer stick has a micro SD, so will just use that.

closed 2018-12-03 by zezba9000

LobizonXP, 2018-12-03:
I think that the Lakka website is really self explanatory, but I will help you anyway:
1- Download your desired Lakka PC image (32 or 64) from here: http://www.lakka.tv/get/windows/generic/
2- Uncompress the image and use Etcher https://www.balena.io/etcher/ to burn it into a USB drive.
3- Boot your PC with that USB drive, and before the system boots, type "installer".
4- The installer will guide you to install Lakka in the desired drive.
5- Enjoy!

NOTE: It will wipe out all the data in the drive.

zezba9000, 2018-12-03:
Tried that, didn't work. So I thought I was missing something. In either case I've resolved the issue a different way.