libretro/Lakka-LibreELEC #572: GameHAT GPIO support via mk_arcade_joystick_rpi

RiRilot, 2018-12-07:
I’ve recently picked up a GameHat: https://www.waveshare.com/game-hat.htm 4 It’s quite a fun little thing but the drivers it uses are based on mk_arcade_joystick_rpi which is GPIO based. https://github.com/recalbox/mk_arcade_joystick_rpi

Could this package be integrated in to Lakka?

Luraktinus, 2019-03-17:
any news on this, i wanna buy that GameHAT

natinusala, 2019-03-26:
I just had a look at what the driver package looks like - it's nothing but a kernel module, `mk_arcade_joystick_rpi`, probably a modified version of https://github.com/recalbox/mk_arcade_joystick_rpi (or the same, I didn't look). 

In addition, there's a dtb overlay and some additional config.txt entries for the HDMI display.

I would add the kernel module as a driver in the kernel directly (as a `linux` package patch), and add an entry to the Lakka wiki to explain how to make the appropriate changes needed to use the display. Finally, I would add an entry to the joypad autoconfig repo to add the automagic mapping.

However while I do own a RPi3B I don't own the Game HAT. I can order one if needed.

natinusala, 2019-03-30:
Status update: driver added to Lakka, everything works, input, audio and video. 

Expect a PR by the end of the week end ;)

RiRilot, 2019-03-31:
Brilliant! Looking forward to testing it.

closed 2019-04-01 by kivutar

RiRilot, 2019-04-07:
Just tested this. Works perfectly. Nice one!

PR #629 (2019-04-01): RPi2: added Game HAT driver and instructions
Fixes #572.

The driver is always built but not enabled by default (because it may collide with any other GPIO related driver or software). 

The config is entirely manual because the driver is generic for all recalbox-compatible GPIO controllers (vid/pid are both set to 1).

Input mapping and instructions can be found in the README.md file. It can be added to the Lakka documentation if wanted.

Merging this PR may also merge (or close) #615.