libretro/Lakka-LibreELEC #577: replace MAME 2003 with MAME 2003-Plus?

markwkidd, 2018-12-19:
I hope that an Issue is the appropriate place to ask this question.

As you all may know, there has been a loose group of five or six coders working on MAME 2003-Plus over the last year. The purposes of this work include:
  * Closer integration with the libretro API
  * Support for approximately 300 new games so far along with some neat hacks such as CD quality soundtrack hacks
  * New content-aware input handling and core option functionality which only presents controls and options that are relevant for the game currently being played.
  * Support for button names in the frontend such as "Strong Kick" instead of generic "Button 3"
  * Bugfixes to current code and backports from later MAME code
  * Keeping a focus on a good experience for low-power devices

On the RetroArch side, there is a scanner RDB database for MAME 2003-Plus and there are thumbnails for it in the RetroArch downloader as part of the MAME pack. From what I can tell, the complete romsets for MAME 2003-Plus are widely available, and about 95% of of MAME 0.78 romsets work as-is with MAME 2003-Plus anyway.

Would you consider switching from MAME 2003 to MAME 2003-Plus as the built-in MAME?

Ntemis, 2018-12-19:
That's the more sane and inevitable thing to be done

5schatten, 2018-12-28:
I use mame2003-plus quite a while now on generic/RPi/AML and it works quite good & since it's actively maintained it's probably a good idea to use it as a default core. But I would wait until [this](https://github.com/libretro/retroarch-assets/issues/263) is resolved to give it a polished look.

Ntemis, 2019-01-21:
@kivutar  imo you should close this issue