libretro/Lakka-LibreELEC #578: MXQ PRO 4k S905x p212 - have various bugs to me

kryspus, 2018-12-20:
I have installed sugested by lakka rom for s905 devices. I dont swap the dtb file for "x" vesrion s905, becouse after i tryed the oryginal one its runned up to RetroArch (GUI interface), so i asume its not needed.
And here is it few bugs i have and i can't fix it by myself.

Cant connect to wi-fi, in GUI interface when i navigate to into Wi-FI setup its showing only WiFi icon with "[]" (a space), when i click "ok" there is showing Phaseprasse to enter.
Wired connection work fine, so I instaled PuTTy and tryed the solution from FAQ, and stuck at this:
connmanctl
Error getting VPN connections: The name net.connman.vpn was not provided by any connmanctl> agent on
Agent registered
The bluetooth conection don't work. I have connected my PS3 controler and its showed up "conected to port #0)"
Lakka:~ # bluetoothctl
[bluetooth]# agent on [WORK]
[bluetooth]# default-agent [WORK]
[bluetooth]# power on
No default controller available [<-- [Problem]
[bluetooth]# discoverable on
No default controller available
[bluetooth]# pairable on
No default controller available
[bluetooth]# scan on
No default controller available
I have placed my commodore games in ROMs folders, and perform scaning, after it finally finished there is no new tabs, but i could bypass this by Loading Core (PS1 games case) and Load Content manually.

Games start but there is no controller who works in the game. L3+R3 work fine for the RetroArch menu, but in game work nothing, i just see the loaded first screen.
In Playstation 1 case i runed Colin and i was able to move in the menu (not tested further)

Due a lot of (fundamental) function is not working for me i wonder do i do it all right and maybe its some other version for my device i should try, or i doing some main thnig wrong? http://le.builds.lakka.tv/S905.arm/?C=M&O=D

Device:
MXQ PRO 4k S905x p212
System information:
Build date: Jun 23 2018
Git: 2a2e732
GCC (7.3.0) 32-bit
NEON VFPv3 VFPv4 ASIMD
ARMv8
unix
Lakka: S905.arm-2.1.1
Linux (v3.14)
RetroRating level : -1
Video content driver: mali-fbdev
...

Sorry for my english, and forgive me if i don't do research i should before writing here.

natinusala, 2018-12-20:
Having `[]` in the networks list means that you don't have any valid wlan interface on your device.

You mentioned the DTB, are you using the correct driver ?

orbea, 2018-12-20:
Given that this is a lakka issue and the new issue is already made for lakka I am going to close this.

Please discuss this further here. https://github.com/libretro/Lakka-LibreELEC/issues/578

kryspus, 2018-12-20:
Thx for reply natinusala.
I have many wLANs around my device, which my laptop can easily see. 

According to this guide: https://mxqproject.com/amlogic-s805-s905-s812-s912-lakka-retroarch-gaming-system/
I miss the steps with replacing dtb file, becouse in official lakka site while I downloaded game and following installation steps in their site there was nothing about replacing dtb file. 
I understand this dtb file is needed for proper boot lakka, so once it booted I assume it's not needed. Do I wrong? Is this file is some kind of driver you telling?

natinusala, 2018-12-20:
The DTB is sure part of the driver, you could try with

Ntemis, 2018-12-20:
change the default dtb with the one you use on libreelec

kryspus, 2018-12-21:
@natinusala 
I have tryed already all the dtb files defignated for 1G device with s905x processor and there is no visible changes, so i thnik this is not the case.
@Ntemis 
I dont use LibreElec, I use custom android TV ROM. Is this ROM could contains such a dtb file i could try replace to Lakka?

Ntemis, 2018-12-21:
No, android dtb is not compatible with linux

kryspus, 2019-01-02:
bump.
How about that bugs?

Ntemis, 2019-01-02:
You must find a compatible dtb for your board.
By default lakka only supports gbit lan only

kryspus, 2019-01-02:
"By default lakka only supports gbit lan boards."
Is that mean lakka by default don't support wi-fi?
So, you telling me all that problems I described is becouse I have wrong dtb file?

kryspus, 2019-01-02:
I,m not a programmer but If I can help you somehow by sending some file I could try.
Thank you for explaining. I really like the idea of lakka, but for me it's not doing their job now. Looking forward then.

closed 2020-08-18 by Ntemis