libretro/Lakka-LibreELEC #579: S905 DTB links in docs point to compromised domain

markwkidd, 2018-12-21:
The Lakka docs for the S905 include links to a site for downloading the necessary DTB file. Here is the docs page I'm referring to: http://www.lakka.tv/doc/S905/

The DTB link points to https://sx05.eu/ which give me a security warning when I click on it in the Chromse browser. When I override the security warning, I see that the site looks to have gone offline.

gouchi, 2019-01-04:
I don't understand links go to [https://kszaq.libreelec.tv/s905/8.2/device_trees/S905/](https://kszaq.libreelec.tv/s905/8.2/device_trees/S905/) on the website.

And there is an explanation why [sx05.eu is no more available](https://forum.libreelec.tv/thread/12432-notice-sx05-eu-domain-for-kszaq-builds-no-longer-in-use/).

ToKe79, 2019-01-05:
I concur with @gouchi - on lakka.tv/docs and also in the wiki here it points to kszaq.libreelec.tv site. Maybe you have cached older version of the sub-page.

closed 2019-01-05 by ToKe79