libretro/Lakka-LibreELEC #581: wifi doesn't work with Bqeel M9Cmax S905X

markwkidd, 2018-12-21:
I have tried Lakka 2.1.1 and the two most recent nightlies. In all cases, the wifi scanner doesn't seem to be working (it displays an empty white box, like a missing Unicode character on a webpage).

I am using a DTB which I am pretty confident is correctly matched for the device and for LibreELEC 8.2. I'm including that file with this post just in case.

My plan is to give this box away as a gift on December 27 but if there is anything I can do to test it between now and then please let me know.

[gxl_p212_2g.zip](https://github.com/libretro/Lakka-LibreELEC/files/2704415/gxl_p212_2g.zip)

markwkidd, 2018-12-24:
I checked and this is still the case with the new Lakka 2.2 release.

jzwtxp, 2018-12-30:
USB wireless network card can work

markwkidd, 2019-03-07:
I'm closing this as a duplicate of https://github.com/libretro/Lakka-LibreELEC/issues/502 but it does still seem to be an issue

closed 2019-03-07 by markwkidd