libretro/Lakka-LibreELEC #582: ShanWan PS3/PC controller rumbling always on

gingerluo, 2018-12-25:
- Which version of Lakka are you using? Lakka-RPi2.arm-2.1.1
- What system hardware are you using?  RaspPi2
- What did you do?   I have a SHANWAN PS3 DualShock controller plugged in USB
- What did you expect to happen?  Don't rumble all the time, even from the time I plugged in.
- What happened instead?  It start rumbling immediate after I plugged in the controller.

$ dmesg
[  534.985552] input: PLAYSTATION(R)3 Controller as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/bluetooth/hci0/hci0:70/0005:054C:0268.0002/input/input1
[  534.987760] sony 0005:054C:0268.0002: input,hidraw1: BLUETOOTH HID v0.00 Joystick [PLAYSTATION(R)3 Controller] on 00:1a:7d:da:71:13
[  557.585281] usb 1-1.2: new full-speed USB device number 6 using dwc_otg
[  557.681430] usb 1-1.2: New USB device found, idVendor=054c, idProduct=0268
[  557.681448] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  557.681457] usb 1-1.2: Product: PS3 GamePad
[  557.681467] usb 1-1.2: Manufacturer: SHANWAN
[  557.697661] input: SHANWAN PS3 GamePad as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:054C:0268.0003/input/input2
[  557.703602] sony 0003:054C:0268.0003: input,hiddev0,hidraw2: USB HID v1.10 Joystick [SHANWAN PS3 GamePad] on usb-3f980000.usb-1.2/input0
[  557.708925] sony 0003:054C:0268.0003: controller with MAC address 01:97:ab:90:63:25 already connected
[  557.756233] sony: probe of 0003:054C:0268.0003 failed with error -17
[  558.660710] usb 1-1.2: USB disconnect, device number 6
[  558.935272] usb 1-1.2: new full-speed USB device number 7 using dwc_otg
[  559.046946] usb 1-1.2: New USB device found, idVendor=054c, idProduct=0268
[  559.046966] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  559.046975] usb 1-1.2: Product: PS3 GamePad
[  559.046984] usb 1-1.2: Manufacturer: SHANWAN
[  559.062477] input: SHANWAN PS3 GamePad as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:054C:0268.0004/input/input3
[  559.063868] sony 0003:054C:0268.0004: input,hiddev0,hidraw1: USB HID v1.10 Joystick [SHANWAN PS3 GamePad] on usb-3f980000.usb-1.2/input0

Lakka:/dev/usb # lsusb
Bus 001 Device 005: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 007: ID 054c:0268 Sony Corp. Batoh Device / PlayStation 3 Controller   // This is the one problematic
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Thanks!!

gingerluo, 2019-02-26:
Anyone?

flihp, 2019-02-27:
I've reproduced this issue on an Intel MinnowboardMax running Lakka 2.2.1. With the controller plugged in (and vibrating constantly) I'm able to pair over bluetooth, then unplug the controller (causing the vibration to stop) and use it as expected. As soon as I plug it back in it starts vibrating again.

Root cause is discussed here: https://askubuntu.com/questions/861895/ps3-controller-shanwan and there seems to be a patch available to fix the issue (available via same link). I have not however tested this patch.

clach04, 2019-03-08:
I get the same behavior with Lakka 2.2 on a Rasppberry pi 3b+ as above (Bluetooth works fine).

I believe Lakka 3 is supposed to be based on libreELEC 9 which I believe will have the later kernel that includes a fix (4.15).

gouchi, 2019-03-08:
Yes it should.

[Source](https://forums.libretro.com/t/shanwan-pc-ps3-android-wireless-controller-on-lakka/8887/9)

msdos, 2019-04-03:
I have the same problem in RPi3. Teste using Lakka and a Retropie last version installation. When using bluetooth, works flawlessly.

coldspark29, 2020-01-31:
Same issue with same controller on 2.3.2 with the Generic PC 64-bit image with no Vulkan on NVidia

gouchi, 2020-02-01:
Lakka 2.3.2 is not running kernel 4.15.

You should try to make a test with future [Lakka 3.0](http://nightly.builds.lakka.tv/Lakka-LE9.2/oven/). Please bear in mind it is a still WIP.

gnif, 2020-02-29:
I suggest you update the kernel, please see: https://github.com/torvalds/linux/commit/69481059c507b3b5a99aaab0d544e6516034defc

closed 2020-08-18 by Ntemis