libretro/Lakka-LibreELEC #585: Default overlay path not set correctly / not using overlayfs

metchebe, 2018-12-28:
System: Lakka 2.2 on Raspberry Pi 3 with default settings, updated from 2.1.1

According to [pull request 410](https://github.com/libretro/Lakka-LibreELEC/pull/410), it seems that the intention is to use the OverlayFS system for the 'overlay' directory, however it is not in use since the mount point `/tmp/overlays` does not exist, at least in my install.

The default retroarch.cfg has the directory set as:

`overlay_directory = "~/.config/retroarch/overlay"`

which is an empty directory (and the default retroarch setting). On the other hand, the overlays shipped with Lakka are in

`/usr/share/retroarch-overlays`

The OverlayFS system is the best solution, but if it does not work then at least the shipped overlays path should be used.

natinusala, 2018-12-28:
Could you use systemctl to check the status of the `tmp-overlays.mount` unit ? This should explain why the overlayfs mount doesn't work for you.

In all cases the default config is wrong and should be fixed.

metchebe, 2018-12-28:
OK, so the unit was disabled. I enabled and started it and it works now, thanks!

Should it have been disabled? I don't have a spare SD card to test a clean flash.

natinusala, 2018-12-28:
Well the unit is clearly missing from https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/libretro/retroarch/package.mk that needs to be fixed

PR #678 (2019-06-14): enable unit for overlays
fixes #585
closed 2019-06-14 by natinusala