libretro/Lakka-LibreELEC #586: lakka 2.2 nightly builds Wi-Fi not working on amlogic s905 devoces

jzwtxp, 2018-12-29:
s905 s905x Wi-Fi not working

jzwtxp, 2018-12-30:
Onboard Wi-Fi not working （rtl8189）

But the onboard wireless network card can work

hojnikb, 2019-01-12:
It's not working for me either from 2.1.1 onwards. 
Also, what is the state of rtl8723bs based S905/x/w devices ? On my TX3 Max with Tanix libreelec dtb wifi works (on 2.0) but bluetooth works only partially. 

For example, i can see one MAC address of my dualshock3 controller, but i can't connect to it. When using an external usb bluetooth, it connects just fine.

markwkidd, 2019-02-21:
Probably the same issue as https://github.com/libretro/Lakka-LibreELEC/issues/581

markwkidd, 2019-03-07:
Also probably the same issue as https://github.com/libretro/Lakka-LibreELEC/issues/502

ToKe79, 2019-09-21:
can you test this build?
http://nightly.builds.lakka.tv/PRs/pr_800_000093/S905.arm/
it is based on latest release + some wifi drivers

closed 2020-08-18 by Ntemis