libretro/Lakka-LibreELEC #589: Asus gamepad not working correclty until lakka restart

jmandawg, 2019-01-06:
Hi guys,

Most times when i startup Lakka and then turn on my Asus gamepad bluetooth controller, it will scroll through the menus crazy, like sometimes only allowing the first and last menu item to be selected, sometimes skipping over items.  If i'm able to make it to the "restart lakka" menu item and restart lakka service (or from ssh), it work fine.  Sometimes it will fix itself,  every once in a while (rarely) it will just work.

lakka log during connect:
```
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] Resetting shader to defaults ... 
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
[INFO] [udev]: Plugged pad: ASUS Gamepad (2821:17664) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports 0 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
```

dmesg during connect:
```
[    9.834019] Bluetooth: Core ver 2.22¬
[    9.834111] NET: Registered protocol family 31¬
[    9.834116] Bluetooth: HCI device and connection manager initialized¬
[    9.834143] Bluetooth: HCI socket layer initialized¬
[    9.834157] Bluetooth: L2CAP socket layer initialized¬
[    9.834192] Bluetooth: SCO socket layer initialized¬
[    9.845460] Bluetooth: HCI UART driver ver 2.3¬
[    9.845476] Bluetooth: HCI UART protocol H4 registered¬
[    9.845482] Bluetooth: HCI UART protocol Three-wire (H5) registered¬
[    9.845672] Bluetooth: HCI UART protocol Broadcom registered¬
[   11.211168] random: crng init done¬
[  174.249605] Bluetooth: HIDP (Human Interface Emulation) ver 1.2¬
[  174.249639] Bluetooth: HIDP socket layer initialized¬
[  174.254426] hid-generic 0005:0B05:4500.0001: unknown main item tag 0x0¬
[  174.255157] input: ASUS Gamepad as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:0B05:4500.000
[  174.259819] hid-generic 0005:0B05:4500.0001: input,hidraw0: BLUETOOTH HID v0.40 Gamepad [ASUS Gamepad] on b8:27:eb:cxxxxx
```

Contents of config file:
```
Lakka:~ # cat joypads/udev/ASUS\ Gamepad.cfg 
input_a_btn = "1"
input_b_btn = "0"
input_device = "ASUS Gamepad"
input_down_btn = "h0down"
input_driver = "udev"
input_l2_axis = "+5"
input_l3_btn = "7"
input_l_btn = "4"
input_l_x_minus_axis = "-0"
input_l_x_plus_axis = "+0"
input_l_y_minus_axis = "-1"
input_l_y_plus_axis = "+1"
input_left_btn = "h0left"
input_r2_axis = "+4"
input_r3_btn = "8"
input_r_btn = "5"
input_r_x_minus_axis = "-2"
input_r_x_plus_axis = "+2"
input_r_y_minus_axis = "-3"
input_r_y_plus_axis = "+3"
input_right_btn = "h0right"
input_select_btn = "9"
input_start_btn = "10"
input_up_btn = "h0up"
input_x_btn = "3"
input_y_btn = "2"
```
I also have the following udev rule:
```
Lakka:~ # cat /etc/udev/rules.d/99-asus-gamepad-controller.rules 
SUBSYSTEM=="input", ATTRS{name}=="ASUS Gamepad", MODE="666", ENV{ID_INPUT_JOYSTICK}="1"

```

gouchi, 2019-01-07:
Hi,

Can you try this [operation](https://forums.libretro.com/t/dualshock-4-crazy-fast-repeating-controls/14197/6) for the asus gamepad autoconfig file ?

Thank you.

jmandawg, 2019-01-08:
Deleting the autoconfig doesn't help.

But tt seem work if i let the controller set for around 10 seconds without touching it after turning it on.

gouchi, 2019-01-21:
Hi,

May we get some [logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/) at least Lakka system log, dmesg.

Thank you.

jmandawg, 2019-01-26:
So this only seems to happen in the lakka UI (I'm using RGUI), once a game starts the controller is fine.
When it is broken, it acts like when you hold up or down for a long time in the RGUI menu and it reaches full acceleration speed, and will only toggle between 2 menu items.  

Last few times i tried, it only acts up when it first connects (will jump through the menu and then acts normal after that)

Here is the lakka log.
```
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
[INFO] [recording] twitch streaming key empty
[INFO] RetroArch 1.7.5 (Git fcf75fd)
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Dec 25 2018
[INFO] Version: 1.7.5
[INFO] Git: fcf75fd
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x1.
[INFO] [GL]: Found GL context: videocore
[INFO] [GL]: Detecting screen resolution 720x480.
[INFO] [EGL]: eglSwapInterval(0)
[INFO] [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] Querying EGL extension: KHR_image => exists
[INFO] [GL]: Using resolution 720x480
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] Resetting shader to defaults ... 
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
[INFO] [udev]: Plugged pad: ASUS Gamepad (2821:17664) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports 0 force feedback effects.
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/pi_iq_horz_nearest_vert.glsl"
[INFO] Found shader "/tmp/shaders/pi_iq_horz_nearest_vert.glslp"
[INFO] Found shader "/tmp/shaders/retroarch.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[... 276 lines cut ...]
[    9.545556] Bluetooth: HCI UART driver ver 2.3
[    9.545571] Bluetooth: HCI UART protocol H4 registered
[    9.545576] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    9.545746] Bluetooth: HCI UART protocol Broadcom registered
[   10.287921] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   10.287940] Bluetooth: HIDP socket layer initialized
[   10.290230] hid-generic 0005:0B05:4500.0001: unknown main item tag 0x0
[   10.290440] input: ASUS Gamepad as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:0B05:4500.0001/input/input0
[   10.291175] hid-generic 0005:0B05:4500.0001: input,hidraw0: BLUETOOTH HID v0.31 Gamepad [ASUS Gamepad] on b8:27:eb:c2:17:2e
[   10.753485] random: crng init done
[   29.239797] hid-generic 0005:0B05:4500.0002: unknown main item tag 0x0
[   29.240459] input: ASUS Gamepad as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12/0005:0B05:4500.0002/input/input1
[   29.242023] hid-generic 0005:0B05:4500.0002: input,hidraw0: BLUETOOTH HID v0.31 Gamepad [ASUS Gamepad] on b8:27:eb:c2:17:2e
[   79.495407] hid-generic 0005:0B05:4500.0003: unknown main item tag 0x0
[   79.495905] input: ASUS Gamepad as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:0B05:4500.0003/input/input2
[   79.497467] hid-generic 0005:0B05:4500.0003: input,hidraw0: BLUETOOTH HID v0.31 Gamepad [ASUS Gamepad] on b8:27:eb:c2:17:2e
[  122.252356] hid-generic 0005:0B05:4500.0004: unknown main item tag 0x0
[  122.252893] input: ASUS Gamepad as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:12/0005:0B05:4500.0004/input/input3
[  122.254187] hid-generic 0005:0B05:4500.0004: input,hidraw0: BLUETOOTH HID v0.31 Gamepad [ASUS Gamepad] on b8:27:eb:c2:17:2e
[  263.639538] hid-generic 0005:0B05:4500.0005: unknown main item tag 0x0
[  263.639977] input: ASUS Gamepad as /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:0B05:4500.0005/input/input4
[  263.641383] hid-generic 0005:0B05:4500.0005: input,hidraw0: BLUETOOTH HID v0.31 Gamepad [ASUS Gamepad] on b8:27:eb:c2:17:2e
[  304.775699] brcmfmac: brcmf_run_escan: error (-25)
[  304.775709] brcmfmac: brcmf_cfg80211_scan: scan error (-25)
`

gouchi, 2019-01-27:
Thank for the information. It seems the gamepad is recognized as keyboard.

Can you try to provide also [those informations](https://github.com/libretro/Lakka-LibreELEC/issues/314#issuecomment-352588951) ?

We may need to add udev rules.

Thank you.

jmandawg, 2019-01-30:
Sorry about the delay, here is the output from the command:
```
Lakka:~ # cat /proc/bus/input/devices 
I: Bus=0005 Vendor=0b05 Product=4500 Version=0040
N: Name="ASUS Gamepad"
P: Phys=b8:27:eb:c2:17:2e
S: Sysfs=/devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:0B05:4500.0001/input/input0
U: Uniq=38:2c:4a:8b:9b:5f
H: Handlers=kbd leds js0 event0 
B: PROP=0
B: EV=2001b
B: KEY=70db0000 0 0 0 1000 40000000 0 0 0 0
B: ABS=700 30627
B: MSC=10
B: LED=f

Lakka:~ # udevadm info --name=input/event0
P: /devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:0B05:4500.0001/input/input0/event0
N: input/event0
E: DEVNAME=/dev/input/event0
E: DEVPATH=/devices/platform/soc/3f201000.serial/tty/ttyAMA0/hci0/hci0:11/0005:0B05:4500.0001/input/input0/event0
E: ID_BUS=bluetooth
E: ID_FOR_SEAT=input-platform-soc
E: ID_INPUT=1
E: ID_INPUT_JOYSTICK=1
E: ID_INPUT_KEY=1
E: ID_PATH=platform-soc
E: ID_PATH_TAG=platform-soc
E: MAJOR=13
E: MINOR=64
E: SUBSYSTEM=input
E: TAGS=:seat:uaccess:
E: USEC_INITIALIZED=15311928

```

And here is a udev rule I had setup (maybe it is wrong):
```
 cat /etc/udev/rules.d/99-asus-gamepad-controller.rules 
#SUBSYSTEM=="input", ATTRS{name}=="ASUS Gamepad", MODE="666", ENV{ID_INPUT_JOYSTICK}="1"

```

gouchi, 2019-01-31:
No problem, thank you for the log.

Please add the file `99-asus-gamepad-controller.rules` in `/storage/.config/udev.rules.d/` with this content

```SUBSYSTEM=="input",ATTRS{name}=="ASUS Gamepad", MODE="666", ENV{ID_INPUT_JOYSTICK}="1" , ENV{ID_INPUT_KEYBOARD}=""```

Then You should reboot or restart udev rules with

```
udevadm control --reload
udevadm trigger
```
Thank you.

jmandawg, 2019-01-31:
Didn't help, but if i disable "Auto Configuration" in the input settings it's doesn't happen as often.  With "auto configuration" turned on the problem is pretty persistent.

jmandawg, 2019-02-14:
I'm going to close this out, it works better with "auto configuration" disabled.

closed 2019-02-14 by jmandawg