libretro/Lakka-LibreELEC #590: Ps4 controller issues

Tom-Neverwinter, 2019-01-08:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 
2.2
- What system hardware are you using?
RPI3
- What did you do?
boot up system... use like normal...
- What did you expect to happen?
controllers to function normally. 
- What happened instead?
PS4 controller model; CUH-ZCT2U has the same issue as before.... scrolls through menu at lightspeed.... 
Model; CUH-ZCT1E is not recognized at all.

gouchi, 2019-01-11:
Hi,

Can you try this operation for the [ps4](https://forums.libretro.com/t/dualshock-4-crazy-fast-repeating-controls/14197/6) gamepad autoconfig file ?

Moreover please provide some log at least [Lakka system and dmesg logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

Tom-Neverwinter, 2019-01-11:
That's not going to work. 

input_vendor_id = "2578"
input_product_id = "1"

however this device reports:
1356 2508

the other model I have which is properly read is; 
1356 1476

doesn't require a log, just requires proper support for the hardware and product id's.

closed 2020-08-18 by Ntemis