libretro/Lakka-LibreELEC #592: SNES Thumbnails don't download

omegaoncoffee, 2019-01-16:
- Which version of Lakka are you using?
2.2.1

- What system hardware are you using?
Raspberry Pi 3 B+

- What did you do?
Downloading thumbnails for snes

- What did you expect to happen?
Complete download

- What happened instead?
Download starts and loads till 7% then it disappears and starts from 0% percent again. Circle repeats 3 times but never completes.

natinusala, 2019-01-16:
It will complete eventually, give it some time

natinusala, 2019-01-16:
It will complete after 15 cycles of 7%, give it some time

omegaoncoffee, 2019-01-16:
Gave it 2 days and several tries. Rather fix a problem not fix around it.

natinusala, 2019-01-16:
This is due to the low memory of the Raspberry Pi but it should still download properly.

The fix is on its way - alfrix is working on separating assets so that you can download individual themes instead of the whole package, reducing the amount of data to download.

omegaoncoffee, 2019-01-16:
Ok thanks. As idea why not split it to part 1, part 2 and so on.

RobLoach, 2019-07-26:
Got reports this is fixed.

closed 2019-07-26 by RobLoach