libretro/Lakka-LibreELEC #596: 8Bitdo NES30 Pro Gamepad stopped working properly

muis24, 2019-01-28:
Since updating to Lakka 2.2, my 8Bitdo NES30 Pro Gamepads stopped working properly, input is repeated and only the up and down buttons work. Manually remapping sort of got them working again, however I am unable to remap the right analogue stick, specifically the x axis. The problem persists in both USB and Bluetooth mode. Running jstest on my Raspberry Pi itself shows that all the buttons and axes work correctly, even the right analogue x (axis 2). I didn't have any issues in Lakka 2.1.1

Logs attached, I can do some more testing if necessary.
[jstest.log](https://github.com/libretro/Lakka-LibreELEC/files/2803866/jstest.log)
[retroarch.log](https://github.com/libretro/Lakka-LibreELEC/files/2803867/retroarch.log)
[dmesg.log](https://github.com/libretro/Lakka-LibreELEC/files/2803868/dmesg.log)

gouchi, 2019-01-31:
Can you make a test with a [nightly build](http://nightly.builds.lakka.tv/) if you reproduce the issue ?

Thank you.

muis24, 2019-02-06:
I tried using Lakka-RPi2.arm-2.2-devel-20190117001852-r28252-gf5f1896.img, unfortunately the issue persist and seems to have regressed even further. On the stable build, I was able to navigate around when using a different wired controller. On the nigthly, plugging in the NES30 basically messes up the whole system, I'm not longer able to navigate, even on the other controller. I tried using 2 different NES 30 controllers. tried all of the 4 usb ports, and 3 different USB wires. Changing to Bluetooth doesn't make any difference. I can confirm that the controllers are still working fine under 2.1.1

ToKe79, 2019-02-06:
How did you switch between the versions? Using the image file in the /storage/.update folder? Or did you boot a "clean" image?

I think there was some regression due to old/obsolete configuration when changing RetroArch version and deleting retroarch.cfg solved the issue. So I would suggest following from shell (typing from my head, don't have my Lakka device available, so adapt as needed):
```
systemctl stop retroarch
mv /storage/.config/retroarch/retroarch.cfg /storage/.config/retroarch.cfg.backup
systemctl start retroarch
```
Maybe it would be not enough and custom joypad mappings would still interfere. Best way to test is to boot a clean image, or renaming the retroarch folder, i.e.:
```
systemctl stop retroarch
mv /storage/.config/retroarch /storage/.config/retroarch.backup
systemctl start retroarch
```

muis24, 2019-02-07:
Thanks for the response.

I used two clean images, I had a spare sd card lying around so I was able to quickly change versions by swapping out the cards, one with the stable 2.1.1 release and one with the nigthly from 20190117, so old or obsolete configs seems unlikely to me. Only thing I did after first boot was turning on SSH so I could connect the controllers with bluetooth and provide some logs.

I tried deleting the configuration folder and restarting retroarch through systemctl. At first, this didn't help. However, as I wanted to get some logs, I stopped retroarch, dmesg'ed en started retroarch manually in verbose mode, and the controller suddenly worked correctly. However after rebooting the controller stops working correctly.

I attached a log containing bluetoothctl, dmesg and retroarch. I'll try again in a few days with the next nightly to try if I can reproduce the issue. My uneducated guess would be that is has something to do with retroarch's input handling and the verbose mode.

[retroarch.log](https://github.com/libretro/Lakka-LibreELEC/files/2840324/retroarch.log)

closed 2020-08-18 by Ntemis