libretro/Lakka-LibreELEC #598: Build segfaults on Debian 9

Zlika, 2019-02-03:
On Debian 9 / AMD64, there is a segfault when compiling openssl:

```
make[2]: Entering directory '/home/thomas/git/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.2-devel/openssl-1.0.2o/.x86_64-libreelec-linux-gnu/test'
make[2]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[2]: Nothing to be done for 'install'.
make[2]: Leaving directory '/home/thomas/git/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.2-devel/openssl-1.0.2o/.x86_64-libreelec-linux-gnu/test'
making install in tools...
make[2]: Entering directory '/home/thomas/git/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.2-devel/openssl-1.0.2o/.x86_64-libreelec-linux-gnu/tools'
make[2]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
make[2]: Leaving directory '/home/thomas/git/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.2-devel/openssl-1.0.2o/.x86_64-libreelec-linux-gnu/tools'
installing libcrypto.a
installing libssl.a
installing libcrypto.so.1.0.0
installing libssl.so.1.0.0
Segmentation fault
make[1]: *** [Makefile:552: install_sw] Error 139
make[1]: Leaving directory '/home/thomas/git/Lakka-LibreELEC/build.Lakka-Generic.x86_64-2.2-devel/openssl-1.0.2o/.x86_64-libreelec-linux-gnu'
Makefile:26 : la recette pour la cible « release » a échouée
make: *** [release] Erreur 2
```

I found a thread on this compilation bug in the Libreelec forum (https://forum.libreelec.tv/thread/12422-openssl-compilation-issue-for-8-2-5-9/), but I don't know how to fix it.

Regards

gouchi, 2019-02-06:
Did you try ?

> You have to install libssl1.0-dev instead of libssl-dev on Debian Stretch

Zlika, 2019-02-06:
Yes. Without success.

bite-your-idols, 2019-03-05:
I have the same issue using ubuntu 16.04 docker container, just for Generic devices, for RPi it builds fine... any solution?

I found another user with the [same issue](https://github.com/libretro/Lakka-LibreELEC/pull/461#issuecomment-387655862)

toolboc, 2019-04-07:
@bite-your-idols, I had the same issue using the included Dockerfile and attempting to build for Generic.x86_64.  I was able to resolve by applying this [modification](https://forum.libreelec.tv/thread/14975-openssl-compilation-issue/?postID=114573#post114573) to the `makeinstall_target()` function in `packages/security/openssl/package.mk`.

bite-your-idols, 2019-04-08:
thanks @toolboc, but finally I moved back to virtualbox so I can not test it, If I give docker another try I will remember this fix.

gouchi, 2019-06-11:
@bite-your-idols or @Zlika your issue should be fixed now, can you confirm it is the case.

Thank you.

ToKe79, 2019-06-12:
Debian (stretch, 9.9) - working.

closed 2019-06-12 by ToKe79