libretro/Lakka-LibreELEC #600: not all systems showing up in menu after loading roms for many systems

pepsi0457, 2019-02-07:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

RobLoach, 2019-02-08:
Provide examples of ones that don't show up. Compare against libretro database.

pepsi0457, 2019-02-08:
Sure... There is no menu for sega cd or psp... On psp I loaded god of war chains of olympus, and sega cd I loaded night trap and sonic cd... The psx menu showed up with only disk 2 of rebel assault 2 when I loaded metal gear solid, rebel assault, and both disks of rebel assault 2. Also missing dreamcast menu for zelda wind waker. All those are in roms directory and were added/scanned. How do I compare with the database?

closed 2020-08-18 by Ntemis