libretro/Lakka-LibreELEC #601: Error compiling

bite-your-idols, 2019-02-08:
I am getting this error when compiling Lakka (latest  master and v2.2) in a Ubuntu server 16.04.5 (64bit) virtual box in my windows 10 64 bit Laptop for RPi2 or for Generic.

any help? 

thanks in advance

```
Parallel mksquashfs: Using 1 processor
Creating 4.0 filesystem on /home/osboxes/Lakka-LibreELEC/target/Lakka-RPi2.arm-2.2-devel-20190208035903-r28252-gf5f1896.system, block size 131072.
[========================================================================================================================================================================/] 32222/32222 100%

Exportable Squashfs 4.0 filesystem, lzo compressed, data block size 131072
        compressed data, compressed metadata, compressed fragments, no xattrs
        duplicates are removed
Filesystem size 347675.81 Kbytes (339.53 Mbytes)
        47.80% of uncompressed filesystem size (727386.15 Kbytes)
Inode table size 348882 bytes (340.71 Kbytes)
        36.07% of uncompressed inode table size (967316 bytes)
Directory table size 484113 bytes (472.77 Kbytes)
        42.49% of uncompressed directory table size (1139264 bytes)
Number of duplicate files found 1707
Number of inodes 29293
Number of files 27994
Number of fragments 1205
Number of symbolic links  526
Number of device nodes 0
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 773
Number of ids (unique uids + gids) 2
Number of uids 1
        root (0)
Number of gids 2
        root (0)
        unknown (81)
Makefile:29: recipe for target 'image' failed
make: *** [image] Error 1
```

natinusala, 2019-02-08:
Are you out of disk space ? Because the build is successful from what I see

bite-your-idols, 2019-02-08:
I don't think so, the virtual image auto-resizes as it compiles/downloads, and in my PC there are around 50GB of free space...

mlnlbrt, 2019-02-08:
You are probably missing `mkimage` binary from `uboot-tools` package.

bite-your-idols, 2019-02-08:
ok, now it works! I compiled Lakka for RPi2... but I get another error when making image for Generic:
`/usr/bin/ld: cannot find /lib64/libm.so.6
/usr/bin/ld: cannot find /usr/lib64/libmvec_nonshared.a
/usr/bin/ld: cannot find /lib64/libmvec.so.1
`

I can find he first and the last of these and create a symlink to that location but there is no libmvec_nonshared.a in my computer... any clue?

thanks a lot

mlnlbrt, 2019-02-08:
No idea, I would start from executing `apt-get` command from https://github.com/libretro/Lakka-LibreELEC/blob/master/Dockerfile.

Please consider to use Docker for building Lakka.

bite-your-idols, 2019-02-08:
interesting.... so using docker I can build Lakka & LibreELEC from windows....

thanks again.

closed 2019-02-09 by bite-your-idols