libretro/Lakka-LibreELEC #605: Lakka 2.2 Ozone Menu Driver, Crash when Show / Hide Advance settings

iokzero, 2019-02-14:
Using Lakka 2.2 

Enable "Show Advance Settings" in order to get to "Driver" Menu in order to change to Ozone Menu Driver. 
When going back to User Interface to switch off "Show Advanced Settings" , the system will crash. 

Switch off however can be done in the config file using a text editor, but need to disable "save setting when turn off retroarch" 

And if you do all this, then going back "User Interface" and enable "Show Advance Setting", the system will crash too. 

Seems like a Ozone Menu Driver issue. 

Sorry about not be able to try out the Night Build, But It would be worth a mention for those Lakka 2.2 stable release user. (maybe a point release ?) Haven't encounter other Ozone menu driver issue just yet (other than known fact that it doesn't support box art, and the box art settings have no effects when using Ozone.)

natinusala, 2019-02-14:
I don't think ozone is supported on Lakka for now, but it should at least be stable and not crash

Do you have access to a terminal (through SSH) ? If so, I can guide you through the steps to diagnose the crash and help me fix it

iokzero, 2019-02-14:
Hi Nathan,

Thanks for your quick response,
Yes, I do have access to a terminal using SSH. It would be great, if you
can guide me to see what might cause this problem.
I assumed Ozone is supported because it is one of the highlight on the
Lakka 2.2 release note.
<http://www.lakka.tv/articles/2018/12/24/lakka-22-christmas-release/>

Overall I think Ozone works great (Other then the problem I mentioned of
course.),
I like the interface, and definitely a great addition to the menu driver
library.

Again thanks for the wonderful work on Lakka.

I.O.K.

natinusala, 2019-02-14:
Oh well I didn't know it was showcased on the release note haha!

However I just realized that regular Lakka doesn't ship with debug symbols... it means that if you successfully get a stack trace from the crash, we won't be able to do anything with it without the debug symbols.

I don't know what to do sadly, apart from providing you with a debug image

iokzero, 2019-02-14:
would the the debug image work on live mode?
or are you able to test it on  your end?

I can try to help test it on the weekend.

I.O.K.

iokzero, 2019-02-14:
Oh forget to mention that, I am using X86_64 version.

I.O.K.

natinusala, 2019-02-14:
What is your device ? Generic ?

Thing is, ozone has evolved since Lakka 2.2 came out, and I would not be surprised if the crash was already solved in RetroArch master

iokzero, 2019-02-14:
I am using the nuc7i7bnh with the X86_64 Lakka 2.2

I am sure Ozone is getting better, and if you think it might not worth a
fix for this version. It is also fair to wait for the next version.
Just want to reported this, just so people know the issue exist (at least
on this platform anyway)

I.O.K.

iokzero, 2019-02-16:
So I did some initial trouble shooting with verbose mode

having Hide Advance setting on / off.
For some reason now turning it back ON seem to work fine with an issue.

but then again turning it OFF crash the system, and the Terminal just says
"Segmentation fault (core dumped)" with no additional information.

I.O.K.

Mr-Jake, 2019-02-17:
Has Lakka been updated to RetroArch 1.7.6?
1.7.6 has the Ozone improvements.

ToKe79, 2019-06-02:
you can try devel builds, there were some updates on retroarch and cores. http://nightly.builds.lakka.tv/

closed 2019-06-13 by ToKe79