libretro/Lakka-LibreELEC #609: Support for Raspberry PI 3 A+

seanmikhaels, 2019-02-26:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

RobLoach, 2019-02-26:
Hmmm? It should work.

seanmikhaels, 2019-02-26:
Trying with latest 2.2 image from get lakka won’t boot. Pop the card into my 3b+ boots fine.  A+ not even the rainbow screen shows.

RobLoach, 2019-02-26:
Strange. I'm assuming your pi has enough power. Do other images work?

seanmikhaels, 2019-02-26:
Very strange indeed, the 3A+ And 3B+ Use the same SOC also using the same power cord as the 3b+ just moving it over to the 3 A+. I’ve tested two other images that work fine, including raspbian & dietpi.

lvii, 2019-04-28:
I have the same problem.

After insert a new lakka flashed TF card in Raspberry PI 3 A+, it show black screen without any output.

When I insert the same TF card into Raspberry PI 3 B+, it could boot into Lakka.

Official raspbian img was ok on Raspberry PI 3 A+.

closed 2019-06-14 by Ntemis