libretro/Lakka-LibreELEC #617: Lakka 2.2.1 image (for Raspberry Pi)

clach04, 2019-03-12:
1. latest download not listed on the website (causes extra upgrade step for new users).
 2. Missing changelog for 2.2.1 since 2.2

I went to download the latest Lakka image from http://www.lakka.tv/get/windows/rpi2/ this leads to http://le.builds.lakka.tv/RPi2.arm/Lakka-RPi2.arm-2.2.img.gz

However after install the download/upgrade option offers 2.2.1 which was not listed on the website. You can see this if the download URL is reduced to http://le.builds.lakka.tv/RPi2.arm/

The Lakka releases https://github.com/libretro/Lakka-LibreELEC/releases does not mention this. Latest is currently https://github.com/libretro/Lakka-LibreELEC/releases/tag/v2.2

gouchi, 2019-03-13:
Can you try to add issue to [Lakka website repository](https://github.com/lakkatv/lakka-website) ?

Thank you.

clach04, 2019-03-14:
@gouchi thanks! I'd not realized that was a repo :-)

Opened https://github.com/lakkatv/lakka-website/issues/67

closed 2019-03-14 by gouchi