libretro/Lakka-LibreELEC #618: Boot loop on old PC (2007 Dell Inspiron 1521) Suspect something with GPU?

limeabeen, 2019-03-13:
Thanks for reading this post. Please correct me if I have posted this in the wrong spot or you need more info.

Tried both 32 and 64 bit generic nightly builds. Both have same error. I tried editing the autostart and config files, but to my surprise when I navigated to them, they didn't exist. I tried to just make my own, but that didn't work in the end.

I was able to boot to the DRAW_USE_LLVM=0 retroarch -v using the live USB.

- Which version of Lakka are you using? 
I believe it is 2.2, but the file I used was Lakka-Generic.x86_64-2.2.img.gz
- What system hardware are you using?
2007 Dell Inspiron 1521, it has an AMD x64 bit CPU (i believe 2 core) with 2GB RAM. it has a 120 GB SSD.
- What did you do?
Installed lakka with the live USB onto SSD.
- What did you expect to happen?
Successful boot of lakka with GUI.
- What happened instead?
Looping of the lakka logo on bootup.

Graphics Card LOGS
[image]

Thanks again for any help!

mlnlbrt, 2019-03-13:
Looks like a duplicate of the following issues:
https://github.com/libretro/Lakka-LibreELEC/issues/312
https://github.com/libretro/Lakka-LibreELEC/issues/322
https://github.com/libretro/Lakka-LibreELEC/issues/486

Please try `systemctl stop retroarch.service && DRAW_USE_LLVM=0 retroarch`

limeabeen, 2019-03-13:
> Looks like a duplicate of the following issues:
> #312
> #322
> #486
> 
> Please try `systemctl stop retroarch.service && DRAW_USE_LLVM=0 retroarch`

Thanks for the reply!
I have tried this, and I am able to get to lakka (blue screen with white text), but not sure how to apply this knowledge to the SSD, and not sure if i'll have to type this in every time?

@mlnlbrt

mlnlbrt, 2019-03-13:
@limeabeen I would try to do something like this:
- copy `/usr/lib/systemd/system/retroarch.service` to `~/.config/system.d/retroarch-llvm-fix.service`
- edit `~/.config/system.d/retroarch-llvm-fix.service`:
add `Environment=DRAW_USE_LLVM=0` in `[Service]` section OR
modify `Execstart=` in `[Service]` section to contain `DRAW_USE_LLVM=0`

- execute `systemctl disable retroarch`
- execute `systemctl enable retroarch-llvm-fix`

I haven't tried it myself so I am not sure it will work, but this should allow you to disable default `retroarch.service` and execute modified one on each boot.
Probably you will need to experiment with this a little bit.

limeabeen, 2019-03-13:
> @limeabeen I would try to do something like this:
>
> * copy `/usr/lib/systemd/system/retroarch.service` to `~/.config/system.d/retroarch-llvm-fix.service`
> * edit [...]

So, I tried what you said, and it worked very well for the USB. I was able to get to lakka and change the fonts and stuff, plus could reboot and let it do its thing (not having to type in commands). However, I still am having issues getting the laptop to boot via the **SSD.** I tried running the installer after doing the above, but still get a constant loop with the lakka fruit logo. Any ideas?
Thanks!! 
[image]
[image]
[image]
@mlnlbrt

mlnlbrt, 2019-03-14:
@limeabeen Good to hear it works. :) 

I guess you need to repeat these steps in your SSD install, have you tried?

limeabeen, 2019-03-14:
@mlnlbrt 

The problem I am having there is that those folders/files don't exist when I navigate to them on the SSD.

mlnlbrt, 2019-03-14:
@limeabeen Do you want to use the whole SSD for Lakka?
If so, then I would suggest to "burn" Lakka.img (using `dd`) straight to SSD instead of USB and boot from the SSD. Lakka will resize its partitions at the first boot to fill the whole disk. 

I use Lakka this way, didn't need to use the installer.

limeabeen, 2019-03-14:
@mlnlbrt 
I'm willing to try that. 

Will this make importing ROMs impossible since lakka would take up the entire SSD? I might need a bit more guidance, as I'm not the best with linux. I understand your message as don't burn the USB using something like Etcher. However, I need clarification on the next part. Do you suggest using dd to "burn" (clone?) the files from the working environment of the USB to the SSD?

mlnlbrt, 2019-03-17:
@limeabeen please try to install Lakka on your SSD using the following build:
http://buildserver3.lakka.tv/Lakka-Generic.x86_64-2.2-devel-20190316194356-r28258-gbd97ec6.img.gz

It will fail at the first run, but after ssh-ing to the SSD install and running the following command it should work fine:
`echo DRAW_USE_LLVM=0 > /storage/.cache/services/retroarch.conf`

Please let me know if it works for you.

limeabeen, 2019-03-17:
@mlnlbrt Im on it.

limeabeen, 2019-03-17:
@mlnlbrt Same issue as before with the loop cycle.

mlnlbrt, 2019-03-17:
@limeabeen Did you executed the command I mentioned in my previous comment in the storage partition of your SSD?

You can directly access command line, please take a look at "Direct access" section below:
http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/

limeabeen, 2019-03-17:
@mlnlbrt I have not executed the command from your previous statement because I cannot access SSH yet. I am reading the direct connection documentation now, however, I am not seeing how to actually get to the command line in order to start typing those commands.

mlnlbrt, 2019-03-17:
@limeabeen This is all about creating `retroarch.conf` file with `DRAW_USE_LLVM=0` inside, in the proper place in storage partition of your SSD Lakka install, so you can try another way: 

boot your laptop from a Live USB linux distro and mount storage partition of your Lakka SSD installation somewhere, then go to `.cache/service` directory and execute `echo DRAW_USE_LLVM=0 > retroarch.conf` there.

limeabeen, 2019-03-17:
@mlnlbrt i am creative a live boot of lubuntu, which I will boot off of. I will mount the SSD to linux lubuntu so that lubuntu can "see" the ssd, then go to .cache/service and run echo DRAW_USE_LLVM=0 > retroarch.conf.

Sound correct?

mlnlbrt, 2019-03-17:
@limeabeen yup, that sounds good. Please note there will be two partition on your SSD Lakka install:
- smaller "system" partition in the beginning of the drive
- "storage" partition filling rest of the drive

You will need to work on the second one

limeabeen, 2019-03-17:
@mlnlbrt lubuntu@lubuntu: /newdrive/.cache/services$ echo DRAW_USE_LLVM=0 > retroarch.conf

Permission Denied.  I tried doing sudo as well. 
(sudo echo DRAW_USE_LLVM=0 > retroarch.conf)
(echo sudo DRAW_USE_LLVM=0 > retroarch.conf)

mlnlbrt, 2019-03-17:
@limeabeen try `sudo su -` first

limeabeen, 2019-03-17:
@mlnlbrt  still permission denied. I am wondering if I mounted the drive correctly as this is the first time i've had to do that...

mlnlbrt, 2019-03-17:
@limeabeen try this way, assuming `/dev/sda2` is storage partition of your SSD Lakka install:
- `sudo su -`
- `cd /tmp && mkdir mountpoint`
- `mount /dev/sda2 /tmp/mountpoint`
- `echo DRAW_USE_LLVM=0 > /tmp/mountpoint/.cache/services/retroarch.conf`

-

limeabeen, 2019-03-17:
@mlnlbrt OK, I think it worked. I do not see any errrors, but no success message either. How do I test?

mlnlbrt, 2019-03-17:
@limeabeen if you want to check if the file was created correctly just check if `cat retroarch.conf` returns `DRAW_USE_LLVM=0`.

If everything was done correctly your Lakka SSD install should work fine now.

limeabeen, 2019-03-17:
@mlnlbrt We did it! It;s booting off SSD, and I am able to restart over and over without any errors. Thank you for all your help! I could not have done it without you!

gouchi, 2019-07-02:
Closing this issue as it is fixed.

closed 2019-07-02 by gouchi

pizdjuk, 2023-12-05:
Still the problem on Samsung N150, Lakka 4.3