libretro/Lakka-LibreELEC #619: Missing firmware for Broadcom AP6255/AP6212 on Z83

marcfehling, 2019-03-14:
A fresh installation of the latest stable release does not list the WiFi/BT module `Broadcom AP6255/AP6212` in my Mini-PC [AZW Z83-II](http://www.azw-online.com/Product/Z83-II-Intel-Mini-PC.html).

Common Linux distributions do not come with the proper firmware according to this [thread](https://askubuntu.com/questions/1007163/ubuntu-16-04-how-do-i-logically-troubleshoot-ap6212-6255-broadcom-wireless-blu). The missing file may be added via
```
wget -O /lib/firmware/brcm/brcmfmac43455-sdio.txt https://github.com/khadas/android_hardware_amlogic_wifi/raw/b6709758755568e4a0ff6e80993be0fc64c77fb9/bcm_ampak/config/6255/nvram.txt
```
However, `LibreELEC` does not allow access to  `/usr/lib` as stated in this [thread](https://forum.libreelec.tv/thread/12970-copy-file-to-lib-firmware-brcm/). A custom build is mandatory to provide the file.

I would like to ask you if you would either add the missing firmware (if legal), or could provide a hint on where to place the missing firmware for a [custom build](http://www.lakka.tv/doc/Compiling-Lakka/).

Meanwhile, I'll give a nightly build try...

marcfehling, 2019-03-14:
Grr, just stumbled over this commit https://github.com/libretro/Lakka-LibreELEC/commit/c3841e579a8e75cafb1586adca66404ba98411cf which was part of the Christmas release, thus should be fixed upstream.

I'll check the nightly build and will let you know about the results. Sorry for any inconvenience.

marcfehling, 2019-03-14:
Nope. Even in the latest nightly build, the Broadcom network adapter is missing in `lspci`.

The required firmware `brcmfmac43455-sdio.txt` is not available in `/usr/lib/firmware/brcm`. The folder is not empty, but this particular file is missing.

So I have to give a custom build a try. Do you know where to place the missing firmware? Would adding the `wget` line from above work, when I place it at the same position as changes were made in https://github.com/libretro/Lakka-LibreELEC/commit/c3841e579a8e75cafb1586adca66404ba98411cf ?

gouchi, 2019-03-20:
We will have to enable [wlan-firmware-aml](https://github.com/libretro/Lakka-LibreELEC/tree/master/packages/linux-firmware/wlan-firmware-aml) for [Generic platform](https://github.com/libretro/Lakka-LibreELEC/blob/master/projects/Generic/options#L94).

I will try to provide testing image ASAP.

Thank you.

gouchi, 2019-03-23:
Hi,

Please try this [testing build](http://nightly.builds.lakka.tv/special_builds/gouchi/619/).

Thank you.

marcfehling, 2019-03-28:
I'll give it a try next week. Sorry for the delay.

marcfehling, 2019-04-20:
I installed your testing build on my device. Unfortunately, there is still no WiFi/BT module available, i.e. none is listed in `lspci` :-(

marcfehling, 2019-04-20:
There's a guy who managed to get WiFi running on this device with Ubuntu and published a [manual](http://liamryandev.blogspot.com/p/installing-ubuntu-mate-on-kodlix-z83-v.html).

I would like to give it a try with a custom build. Do you have a hint on how to supply individual firmware?

marcfehling, 2019-04-20:
`/lib/firmware/brcm/` contains several files related to `bcm43455`, but none is related to the [SDIO](https://wiki.freebsd.org/SDIO) device that seems to be installed. `/sys/firmware/efi/efivars/` is empty.
```
# find /lib/firmware/brcm/ -name "*43455*"
/lib/firmware/brcm/fw_bcm43455c0_ag.bin
/lib/firmware/brcm/fw_bcm43455c0_ag_apsta.bin
/lib/firmware/brcm/fw_bcm43455c0_ag_p2p.bin
```
---

I took a closer look on this [thread](https://askubuntu.com/questions/1007163/ubuntu-16-04-how-do-i-logically-troubleshoot-ap6212-6255-broadcom-wireless-blu). The author had similar issues with ubuntu running on kernel `4.13`. His `dmesg` output differs from mine.
```
# dmesg | grep -i sdio
[    0.990906] mmc1: new ultra high speed SDR104 SDIO card at address 0001
```
The custom build of Lakka you provided runs on kernel `4.11.12-rt4`. Maybe this issue has been addressed for a more recent kernel? Would an update to the latest `LibreELEC` release be worth a try?

marcfehling, 2019-04-20:
I may try out your other custom build based on LE 9 that was related to a different issue https://github.com/libretro/Lakka-LibreELEC/issues/631#issuecomment-479853902.

marcfehling, 2019-04-21:
Hooray! The custom build based on LE 9 did the trick! Correct firmware seems to be loaded, and there are wireless networks showing up in the XMB menu. I'll check whether Bluetooth controllers connect soon.
```
# dmesg | grep "brcm"
[   10.653962] bluetooth hci0: Direct firmware load for brcm/BCM4345C0.hcd failed with error -2
[   10.653971] Bluetooth: hci0: BCM: Patch brcm/BCM4345C0.hcd not found
[   12.723950] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   12.724113] usbcore: registered new interface driver brcmfmac
[   12.838390] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   12.838490] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.clm_blob failed with error -2
[   12.838498] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   12.838928] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
```
Even audio seems to be supported now.
```
[    2.375947] [drm] HDaudio controller not detected, using LPE audio instead
[    2.507736]   No soundcards found.
[   10.419948] input: Intel HDMI/DP LPE Audio HDMI/DP,pcm=0 as /devices/pci0000:00/0000:00:02.0/hdmi-lpe-audio/sound/card0/input6
[   10.420132] input: Intel HDMI/DP LPE Audio HDMI/DP,pcm=1 as /devices/pci0000:00/0000:00:02.0/hdmi-lpe-audio/sound/card0/input7
[   10.420264] input: Intel HDMI/DP LPE Audio HDMI/DP,pcm=2 as /devices/pci0000:00/0000:00:02.0/hdmi-lpe-audio/sound/card0/input8
[   10.804090] input: bytcr-rt5651 Headset as /devices/pci0000:00/808622A8:00/bytcr_rt5651/sound/card1/input9
```
I'll let you know once I performed some tests on WiFi/BT (and audio).

marcfehling, 2019-04-23:
WiFi/BT works. The kernel update did the trick :)

I connected a DS4 controller via BT. However, inputs were duplicated once wireless. I guess one is able to fix this following this [manual](https://forums.libretro.com/t/dualshock-4-crazy-fast-repeating-controls/14197/6). 

Audio via HDMI works when I pick `hdmi:CARD=HDMI,DEV=2` as the audio device.

However, the output via the audio jack does not. There is no sound ouput via `default:bytcrrt5651` or `sysdefault:bytcrtz5651` as audio devices, and I cannot change the audio device automatically via the GUI anymore after a reboot. I'll investigate further and would open a different issue, since this has nothing to do with the original topic.

Are you planning to use LE9 for future releases?

gouchi, 2019-04-23:
Thank you for the feedback.

Maybe it needs a kernel more up to date than the one we are currently using.

Yes, we will use LE9 for Lakka version 3 as per this [ticket](https://github.com/libretro/Lakka-LibreELEC/issues/556).

I will close this issue as it is fixed using Lakka based on LE9.

closed 2019-04-23 by gouchi