libretro/Lakka-LibreELEC #620: Configurations not save as expected

tonycstech, 2019-03-14:
#1 i flash latest build
#2 i enable [Show Advanced Settings]
#3 I load my own [Configuration File] from USB stick.
System loads my config file and everything seems OK.
#4 I reboot the system and configurations are lost.
#5 I repeat loading my configuration file from USB stick.
System loads my config file and everything seems OK.
#6 I [Save Current Configuration]  (It saves one i opened form USB)
#7 i reboot. Configurations are lost.

Expected behavior is: 
Do not reset configurations on reboot ?
Changing config folder should load config files from there after reboot but it does not.
Same problem with every configuration i make. They dont seem to be either stored or restored after reboot.

ToKe79, 2019-06-01:
AFAIK (I think @natinusala  mentioned this on discord) there is no clean exit of RetroArch process when the Reboot / Shutdown menu options are used, hence the settings are not saved (and also causes file system corruption in specific cases?).
To workaround your issue, enable display of the Quit option (if not already enabled) and quit RetroArch. systemd shout restart the RetroArch and check if your settings were kept. If so, then you can shutdown/reboot your device.

natinusala, 2019-06-01:
It can be "fixed" by adding a `&` to the shutdown and reboot commands in RetroArch (`system(...);`).

PR #653 (2019-06-06): retroarch: patch (hack) to allow graceful shutdown
fixes #620 and possible filesystem corruptions
commit 8b22eefe (2019-06-06): retroarch: patch (hack) to allow graceful shutdown (#653)

fixes #620 and possible filesystem corruptions
closed 2019-06-06 by natinusala