libretro/Lakka-LibreELEC #622: Missing version/tag/release/changelog for Lakka 2.2.1 and 2.2.2

clach04, 2019-03-16:
Related to https://github.com/libretro/Lakka-LibreELEC/issues/617

Version 2.2.2 is available (and also 2.2.1), there is no version tag so its not possible to determine what is new.  The latest release listed at https://github.com/libretro/Lakka-LibreELEC/releases and https://github.com/libretro/Lakka-LibreELEC/tags is 2.2

ToKe79, 2019-07-02:
tags added

closed 2019-07-02 by ToKe79

clach04, 2019-07-03:
Thanks! For my ref:

  * https://github.com/libretro/Lakka-LibreELEC/releases/tag/v2.2.1
      * changelog https://github.com/libretro/Lakka-LibreELEC/compare/v2.2...v2.2.1
  * https://github.com/libretro/Lakka-LibreELEC/releases/tag/v2.2.2
      * changelog https://github.com/libretro/Lakka-LibreELEC/compare/v2.2.1...v2.2.2