libretro/Lakka-LibreELEC #625: Xbox One S controller doesn't work on RPi3 through bluetooth

mateusfccp, 2019-03-20:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
2.2.2

- What system hardware are you using?
Raspberry Pi 3B+

- What did you do?
Fresh install of Lakka 2.2.2 on a SD Card.
Boot it.
Access through SSH.
Disable ERTM so I can connect the controller through Bluetooth.
Trust, pair and connect my controller through Bluetooth.
Try to use the controller.

- What did you expect to happen?
The inputs should work as expected, allowing me to navigate through the menu and play games.

- What happened instead?
The left-right buttons were the only ones that worked as expected. All others didn't. The behavior is similar to what is described on #528.
I then tested my controller input through `jstest` and everything outputted as expected, i.e. the inputs are correct, but the behavior not.
Logs are here: https://pastebin.com/XYBS2CG6

mateusfccp, 2019-03-30:
I think the problem may be solved by simply including `xpadneo` on Lakka, as the other drivers don't work very well with Bluetooth on One S controller.

Kodi on Raspbian also don't work well with this controller, and I managed to solve it by compiling and installing `xpadneo`.

`xpadneo` maintainer is trying to mainline it (https://github.com/atar-axis/xpadneo/issues/44), but I don't think it's going to happen too soon.

willemsegd, 2019-10-04:
Hi, pretty new to this... but I've been struggling with the same issue. I have xpadneo running on a buster image with steamlink, works like a charm. Trying to get the same thing runner with lakka.

I couldn't find any "manual" way of getting xpadneo installed with lakka. Do you know if this is even possible? Or would the only solution here be to wait for xpadneo to eventually (maybe? hopefully?) be mainlined?

closed 2020-08-18 by Ntemis

mateusfccp, 2020-08-19:
@Ntemis Has this been solved?

Ntemis, 2020-08-19:
Version 2.x is being decommissioned , no more fixes. Please try latest nightly. Come to our discord channel and you will see the link