libretro/Lakka-LibreELEC #627: No Audio Odroid XU4

cnurre, 2019-03-24:
- Which version of Lakka are you using? 
2.2.2

- What system hardware are you using?
Odroid XU4

Audio was working in version 2.0, but upgrading has stopped all audio. I've tried a number of different HDMI and TV combos. The `aplay -L` returns `null Discard all samples (playback) or generate zero samples (capture)`

I captured some logs that I believe may identify the problem. 

```
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:5033:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/pcm/pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM default
[ERROR] ALSA: Failed to initialize...
[ERROR] Failed to initialize audio driver. Will continue without audio.
```

For completeness, here is the entire log:

```
[INFO] RetroArch 1.7.6 (Git ed5bd80)
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Mar 10 2019
[INFO] Version: 1.7.6
[INFO] Git: ed5bd80
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 1 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 11 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 1: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 2: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 3: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 4: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 5: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 6: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 7: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 8: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 9: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 10: (640x480) 640 x 480, 60 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0xb322a558.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GL]: Vendor: ARM, Renderer: Mali-T628.
[INFO] [GL]: Version: OpenGL ES 3.1 v1.r12p0-04rel0.f9ea82e6bf7f0bb7544260636f375425.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] Setting up menu pipeline shaders for XMB ... 
[INFO] [GLSL]: Compiling ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling simple ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling snow shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling modern snow shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling bokeh shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling snowflake shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] Resetting shader to defaults ... 
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
[INFO] [udev]: Plugged pad: SNES30             SNES30 Joy     (11720:43808) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports 0 force feedback effects.
[INFO] [Autoconf]: 266 profiles found.
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:5033:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib /home/kivutar/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/pcm/pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM default
[ERROR] ALSA: Failed to initialize...
[ERROR] Failed to initialize audio driver. Will continue without audio.
[INFO] [Menu]: Found menu display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [autoconf]: selected configuration: /tmp/joypads/udev/8Bitdo_SN30_USB.cfg
[INFO] [LED]: LED driver = 'null' 0x2bdfe4
[INFO] [MIDI]: Initializing ...
[INFO] [MIDI]: Input disabled.
[INFO] [MIDI]: Output disabled.
[INFO] [MIDI]: Initialized "alsa" driver.
[WARN] Input device ID 5 is unknown to this libretro implementation. Using RETRO_DEVICE_JOYPAD.
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
```

Please let me know if I can help to capture any other info that may be helpful.

gouchi, 2019-05-04:
Hi,

Do you reproduce the issue with [nightly build](http://nightly.builds.lakka.tv/) ?

Thank you.

cnurre, 2019-05-06:
I just tried installing the April 2nd nightly and got the same end result. The only difference I saw was the log has a different path associated with the error. 

```
ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/conf.c:5033:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/pcm/pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM default
[ERROR] ALSA: Failed to initialize...
[ERROR] Failed to initialize audio driver. Will continue without audio.
```

I also tried building from source and got the same errors only with a path that was relative to the machine I was building on. I dug a bit into this but couldn't figure out where it was getting these paths from.

I've also tried installing via the update directory and a clean flash and got the same results. Let me know if you'd like to try any other experiments.

cnurre, 2019-06-08:
I just tried the nightly from May 28 on my XU4 and I'm still seeing similar errors as I've documented previously. Here's the latest capture of the errors.

```
Jun 08 11:15:14 Lakka retroarch[410]: ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/confmisc.c:767:(parse_card) cannot find card '0'
Jun 08 11:15:14 Lakka retroarch[410]: ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
Jun 08 11:15:14 Lakka retroarch[410]: ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/confmisc.c:392:(snd_func_concat) error evaluating strings
Jun 08 11:15:14 Lakka retroarch[410]: ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
Jun 08 11:15:14 Lakka retroarch[410]: ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/confmisc.c:1246:(snd_func_refer) error evaluating name
Jun 08 11:15:14 Lakka retroarch[410]: ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Jun 08 11:15:14 Lakka retroarch[410]: ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/conf.c:5033:(snd_config_expand) Evaluate error: No such file or directory
Jun 08 11:15:14 Lakka retroarch[410]: ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm/alsa-lib-1.1.4.1/src/pcm/pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM default
```

My gut still says this is some sort of build configuration issue since it appears to be looking for files in build directories, but I haven't dug any deeper to better diagnose the root cause of the issue. 

- Chris

gouchi, 2019-06-11:
Thank you for your tests.

The [kernel config](https://github.com/libretro/Lakka-LibreELEC/blob/master/projects/OdroidXU3/linux/linux.arm.conf#L3937) has the module enabled for audio support for Odroid XU3/XU4.

`aplay -L` and `dmesg | grep snd` give nothing ?

Ntemis, 2019-06-11:
The kernel has been updated, can you please test if i upload an image for you?

cnurre, 2019-06-11:
@gouchi 

`aplay -L` returns:
```
null
    Discard all samples (playback) or generate zero samples (capture)
```
and `dmesg | grep snd`

```
[    4.040820] odroid-audio sound: snd_soc_register_card() failed: -1
```

---
@Ntemis I'd be happy to test out an image. Let me know where you upload it and I'll test it tonight.

Ntemis, 2019-06-12:
Please test this image thanks
https://www.androidfilehost.com/?fid=6006931924117893548

cnurre, 2019-06-12:
@Ntemis Unfortunately I got similar errors. I got the same returned values for `aplay -L` and `dmesg | grep snd` as above. I've copied what I believe are the related logs of the errors as well:

```
ALSA lib /home/demetris/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib /home/demetris/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib /home/demetris/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib /home/demetris/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib /home/demetris/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib /home/demetris/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/conf.c:4554:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib /home/demetris/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/conf.c:5033:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib /home/demetris/Lakka-LibreELEC/build.Lakka-OdroidXU3.arm-2.2-devel/alsa-lib-1.1.4.1/src/pcm/pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM default
[ERROR] ALSA: Failed to initialize...
```

Ntemis, 2019-06-13:
Please test this new image too
https://www.androidfilehost.com/?fid=6006931924117894778
Thank you for your time

cnurre, 2019-06-13:
I tried an update and a fresh install and

`aplay -L` still returns:
```
null
    Discard all samples (playback) or generate zero samples (capture)
```
and `dmesg | grep snd` still returns something very similar to previous attempts

```
[    4.257445] odroid-audio sound: snd_soc_register_card() failed: -1
```

Thanks for working with me on this. I'm happy to keep testing!

Here is the full log from `systemctl stop retroarch.service ; retroarch -v`:
```
[INFO] RetroArch 1.7.7 (Git 3064d0d)
[INFO] === Build =======================================
[INFO] Capabilities: NEON VFPv3 VFPv4 
[INFO] Built: Jun 10 2019
[INFO] Version: 1.7.7
[INFO] Git: 3064d0d
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 1 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 11 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 1: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 2: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 3: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 4: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 5: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 6: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 7: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 8: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 9: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 10: (640x480) 640 x 480, 60 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0xb322a558.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GL]: Vendor: ARM, Renderer: Mali-T628.
[INFO] [GL]: Version: OpenGL ES 3.1 v1.r12p0-04rel0.f9ea82e6bf7f0bb7544260636f375425.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] Setting up menu pipeline shaders for XMB ... 
[INFO] [GLSL]: Compiling ribbon shader..
[... 43 lines cut ...]
[INFO] [Autoconf]: selected configuration: /tmp/joypads/udev/8Bitdo_SN30_USB.cfg
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Menu]: Found menu display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x1b9e84
[INFO] [MIDI]: Initializing ...
[INFO] [MIDI]: Input disabled.
[INFO] [MIDI]: Output disabled.
[INFO] [MIDI]: Initialized "alsa" driver.
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[ERROR] failed_to_start_audio_driver
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
```

Build Info:
`Lakka (community): devel-20190613225852-r28294-g996372e (OdroidXU3.arm)`

Ntemis, 2019-06-14:
Are you sure there is nothing wrong with your board? Please try to flash an official image to test. Let me know. Thanks

Ntemis, 2019-06-14:
Look man i consider this bug fix on my part, if you continue having audio issues please RMA the board to hardkernel

closed 2019-06-14 by Ntemis

cnurre, 2019-06-14:
I almost certain it isn't a hardware issue, but I'll check try out previously versions to get more info. I cannot recall offhand which version introduced the issue (I believe somewhere after 2.0) but I remember switching back and forth a couple times where the previous version worked correctly. Once I figure out the exact version I'll open up another issue.