libretro/Lakka-LibreELEC #628: MIDI not enabled?

anothername99, 2019-03-25:
When I go into MIDI settings to select a MIDI Out device, I can't select any device. On Windows and desktop Linux it works fine.

gouchi, 2019-03-27:
We will have to enable [fluidsynth package](https://github.com/libretro/Lakka-LibreELEC/tree/master/packages/audio/fluidsynth).

anothername99, 2019-03-27:
Fluidsynth is a software synthesizer. While adding it would be a nice touch, I am pretty sure that this is unrelated. I have MIDI hardware connected through USB MIDI and it works fine in Windows and desktop Linux without Fluidsynth.

gouchi, 2019-03-28:
I see I was thinking playback directly on the PC.

Did you try to switch audio driver to Alsa not Alsa threaded ?

Thank you.

anothername99, 2019-03-28:
I've tried the audio driver set to ALSA thread, ALSA and null. All give the same result. I am guessing that either the kernel is missing something (perhaps MIDI drivers were stripped out to make LibreELEC/Lakka slimmer?) or ALSA is compiled wrong.

gouchi, 2019-04-06:
May you try this [image](http://nightly.builds.lakka.tv/special_builds/gouchi/628/Lakka-Generic.i386-2.2-devel-20190406193603-r28257-gd51e47b.img.gz) ?

If it is still not working try to load those modules
`modprobe snd-seq-device && modprobe snd-seq-midi`

Thank you.

anothername99, 2019-04-07:
With your image, I can now select MIDI IN and MIDI OUT devices. Thank you.

gouchi, 2019-04-07:
Thank you I am glad it is working.

You did have to modprobe the module right ?

We will have to enable for the kernel at least
```
CONFIG_SND_SEQUENCER=m
CONFIG_SND_RAWMIDI_SEQ=m
```

anothername99, 2019-04-07:
I didn't touch any commands. I wouldn't know how to enter commands in Lakka.

Maybe I should test that playback is actually working (I only saw that selecting the device worked). Could you put in the latest np2kai core? It has MIDI support and would be easy to test. I tried transplanting it into my Lakka setup from a binary from the buildbot but the core wouldn't load.

gouchi, 2019-04-08:
Can you try with with this [np2kai core build](https://transfer.sh/pbPyT/np2kai_libretro.so) ?

Thank you.

anothername99, 2019-04-08:
Your np2kai build loads and MIDI works in the games that I tried. Thank you.

vol-2, 2020-01-16:
Is it yet possible to enable MT-32 support in Lakka? Preferably via Munt?

closed 2020-08-18 by Ntemis