libretro/Lakka-LibreELEC #631: No HDMI audio on Radeon RX 580

TMueller83, 2019-04-04:
On Lakka 2.2.2-generic.x86_64 audio output via HDMI is not working with my Radeon RX 580 graphics card. I have tried all possible HDMI devices (0-5) without success. I also tried the nightly build from 2019-04-02 again without success.

I have tried to play sound on the HDMI devices directly using aplay but there was no sound output on any HDMI device.

I assume the problem is related to the drivers provided with the Linux kernel 4.11. I can confirm, that using Ubuntu 16.04 with kernel 4.15.0-47-generic the HDMI audio output works flawlessly on the RX 580.

gouchi, 2019-04-04:
Can you try this [testing image](https://drive.google.com/drive/folders/128MaH09bjVcB2pR4Kj2XxdBabR-oDP75) based on LE 9 ?

Thank you.

TMueller83, 2019-04-04:
Thank you very much for the testing image. Sadly, the HDMI audio output does not work with this image as well. I have also tried to use aplay to output sound via HDMI without success. Audio output only works via card 2 (see output of aplay -l below) and possibly via card 0, which I did not test.

Below you find the output of aplay -l and aplay -L

```
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: VT2020 Analog [VT2020 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 1: VT2020 Digital [VT2020 Digital]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: PCH [HDA Intel PCH], device 2: VT2020 Alt Analog [VT2020 Alt Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 11: HDMI 5 [HDMI 5]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: DGX [Xonar DGX], device 0: Multichannel [Multichannel]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: DGX [Xonar DGX], device 1: Digital [Digital]
  Subdevices: 1/1
Subdevice #0: subdevice #0
```
```
aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=PCH
    HDA Intel PCH, VT2020 Analog
    Default Audio Device
sysdefault:CARD=PCH
    HDA Intel PCH, VT2020 Analog
    Default Audio Device
front:CARD=PCH,DEV=0
    HDA Intel PCH, VT2020 Analog
    Front speakers
surround21:CARD=PCH,DEV=0
    HDA Intel PCH, VT2020 Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=PCH,DEV=0
    HDA Intel PCH, VT2020 Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=PCH,DEV=0
    HDA Intel PCH, VT2020 Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=PCH,DEV=0
    HDA Intel PCH, VT2020 Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=PCH,DEV=0
    HDA Intel PCH, VT2020 Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=PCH,DEV=0
    HDA Intel PCH, VT2020 Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=PCH,DEV=0
    HDA Intel PCH, VT2020 Digital
    IEC958 (S/PDIF) Digital Audio Output
hdmi:CARD=HDMI,DEV=0
    HDA ATI HDMI, HDMI 0
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=1
    HDA ATI HDMI, HDMI 1
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=2
    HDA ATI HDMI, HDMI 2
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 3
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=4
    HDA ATI HDMI, HDMI 4
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=5
    HDA ATI HDMI, HDMI 5
    HDMI Audio Output
default:CARD=DGX
    Xonar DGX, Multichannel
    Default Audio Device
sysdefault:CARD=DGX
    Xonar DGX, Multichannel
    Default Audio Device
front:CARD=DGX,DEV=0
    Xonar DGX, Multichannel
    Front speakers
surround21:CARD=DGX,DEV=0
    Xonar DGX, Multichannel
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=DGX,DEV=0
    Xonar DGX, Multichannel
    4.0 Surround output to Front and Rear speakers
surround41:CARD=DGX,DEV=0
    Xonar DGX, Multichannel
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=DGX,DEV=0
    Xonar DGX, Multichannel
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=DGX,DEV=0
    Xonar DGX, Multichannel
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=DGX,DEV=0
    Xonar DGX, Multichannel
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=DGX,DEV=0
    Xonar DGX, Multichannel
    IEC958 (S/PDIF) Digital Audio Output
```

gouchi, 2019-04-06:
Thank you for the test.

Is it possible to get the output from

`lspci -nnk | grep -A3 VGA`

and

`dmesg`

With the testing image based on LE9.

Thank you.

TMueller83, 2019-04-09:
Sorry for the delayed answer. Here comes the requested output:
```
lspci -nnk | grep -A3 VGA
01:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere [Radeon RX 470/480/570/570X/580/580X] [1002:67df] (rev e7)
	Subsystem: Sapphire Technology Limited Ellesmere [Radeon RX 470/480/570/570X/580/580X] (Nitro+ Radeon RX 580 4GB) [1da2:e366]
	Kernel driver in use: amdgpu
01:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere [Radeon RX 580] [1002:aaf0]
```
```
dmesg
[    0.019999] tsc: PIT calibration matches HPET. 1 loops
[    0.019999] tsc: Detected 3503.453 MHz processor
[    0.000007] clocksource: tsc-early: mask: 0xffffffffffffffff max_cycles: 0x32800eaf1c4, max_idle_ns: 440795250577 ns
[    0.000015] Calibrating delay loop (skipped), value calculated using timer frequency.. 7009.86 BogoMIPS (lpj=11678176)
[    0.000016] pid_max: default: 32768 minimum: 301
[    0.002575] Dentry cache hash table entries: 2097152 (order: 12, 16777216 bytes)
[    0.003632] Inode-cache hash table entries: 1048576 (order: 11, 8388608 bytes)
[    0.003671] Mount-cache hash table entries: 32768 (order: 6, 262144 bytes)
[    0.003703] Mountpoint-cache hash table entries: 32768 (order: 6, 262144 bytes)
[    0.003841] ENERGY_PERF_BIAS: Set to 'normal', was 'performance'
[    0.003841] ENERGY_PERF_BIAS: View and update with x86_energy_perf_policy(8)
[    0.003844] process: using mwait in idle threads
[    0.003846] Last level iTLB entries: 4KB 512, 2MB 8, 4MB 8
[    0.003846] Last level dTLB entries: 4KB 512, 2MB 32, 4MB 32, 1GB 0
[    0.003848] Spectre V2 : Mitigation: Full generic retpoline
[    0.003848] Spectre V2 : Spectre v2 / SpectreRSB mitigation: Filling RSB on context switch
[    0.003849] Spectre V2 : Enabling Restricted Speculation for firmware calls
[    0.003855] Spectre V2 : mitigation: Enabling conditional Indirect Branch Prediction Barrier
[    0.003856] Spectre V2 : User space: Mitigation: STIBP via seccomp and prctl
[    0.003857] Speculative Store Bypass: Mitigation: Speculative Store Bypass disabled via prctl and seccomp
[    0.004017] Freeing SMP alternatives memory: 52K
[    0.013093] TSC deadline timer enabled
[    0.013095] smpboot: CPU0: Intel(R) Core(TM) i7-3770K CPU @ 3.50GHz (family: 0x6, model: 0x3a, stepping: 0x9)
[    0.013155] Performance Events: PEBS fmt1+, IvyBridge events, 16-deep LBR, full-width counters, Intel PMU driver.
[    0.013172] ... version:                3
[    0.013172] ... bit width:              48
[    0.013172] ... generic registers:      4
[    0.013173] ... value mask:             0000ffffffffffff
[    0.013173] ... max period:             00007fffffffffff
[    0.013174] ... fixed-purpose events:   3
[    0.013174] ... event mask:             000000070000000f
[    0.013200] rcu: Hierarchical SRCU implementation.
[    0.013341] NMI watchdog: Enabled. Permanently consumes one hw-PMU counter.
[    0.013341] smp: Bringing up secondary CPUs ...
[    0.013341] x86: Booting SMP configuration:
[    0.013341] .... node  #0, CPUs:      #1 #2 #3 #4 #5 #6 #7
[    0.025690] smp: Brought up 1 node, 8 CPUs
[    0.025690] smpboot: Max logical packages: 1
[    0.025690] smpboot: Total of 8 processors activated (56077.88 BogoMIPS)
[    0.026860] devtmpfs: initialized
[    0.026860] PM: Registering ACPI NVS region [mem 0xde15d000-0xde1fbfff] (651264 bytes)
[    0.026860] PM: Registering ACPI NVS region [mem 0xdea29000-0xdea6bfff] (274432 bytes)
[    0.026860] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6370867519511994 ns
[    0.026860] futex hash table entries: 2048 (order: 5, 131072 bytes)
[    0.026860] xor: automatically using best checksumming function   avx       
[    0.026860] pinctrl core: initialized pinctrl subsystem
[    0.026951] NET: Registered protocol family 16
[    0.027098] cpuidle: using governor ladder
[    0.027101] cpuidle: using governor menu
[    0.027170] ACPI FADT declares the system doesn't support PCIe ASPM, so disable it
[    0.027171] ACPI: bus type PCI registered
[    0.027212] PCI: MMCONFIG for domain 0000 [bus 00-3f] at [mem 0xf8000000-0xfbffffff] (base 0xf8000000)
[    0.027214] PCI: MMCONFIG at [mem 0xf8000000-0xfbffffff] reserved in E820
[    0.027219] pmd_set_huge: Cannot satisfy [mem 0xf8000000-0xf8200000] with a huge-page mapping due to MTRR override.
[    0.027276] PCI: Using configuration type 1 for base access
[    0.027343] core: PMU erratum BJ122, BV98, HSD29 worked around, HT is on
[    0.086676] raid6: sse2x1   gen() 10721 MB/s
[    0.143343] raid6: sse2x1   xor()  7703 MB/s
[    0.200009] raid6: sse2x2   gen() 12751 MB/s
[    0.256678] raid6: sse2x2   xor()  8554 MB/s
[... 670 lines cut ...]
[   18.212741] snd_hda_codec_via hdaudioC1D2:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[   18.212743] snd_hda_codec_via hdaudioC1D2:    hp_outs=1 (0x28/0x0/0x0/0x0/0x0)
[   18.212744] snd_hda_codec_via hdaudioC1D2:    mono: mono_out=0x0
[   18.212746] snd_hda_codec_via hdaudioC1D2:    dig-out=0x2d/0x2e
[   18.212747] snd_hda_codec_via hdaudioC1D2:    inputs:
[   18.212749] snd_hda_codec_via hdaudioC1D2:      Front Mic=0x29
[   18.212751] snd_hda_codec_via hdaudioC1D2:      Rear Mic=0x2b
[   18.212752] snd_hda_codec_via hdaudioC1D2:      Line=0x2a
[   18.220555] intel_rapl: Found RAPL domain package
[   18.220558] intel_rapl: Found RAPL domain core
[   18.223316] atl1c 0000:07:00.0: atl1c: eth0 NIC Link is Up<1000 Mbps Full Duplex>
[... cut ...]
[   18.226263] input: HDA Intel PCH Front Headphone as /devices/pci0000:00/0000:00:1b.0/sound/card1/input19
[   18.244691] 8021q: 802.1Q VLAN Support v1.8
[   18.328734] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: (null)
[   18.544065] EXT4-fs (sdc2): mounted filesystem with ordered data mode. Opts: (null)
[   18.651661] fuse init (API version 7.27)
[   18.836621] EXT4-fs (sdb6): mounted filesystem with ordered data mode. Opts: (null)
[   23.948754] logitech-hidpp-device 0003:046D:2011.0004: HID++ 1.0 device connected.
```

gouchi, 2019-04-23:
Hi,

Thank you for the log. It does not seem to have any issue.

If you did edit the configuration file directly you did stop RA before as mentioned in the [documentation](http://www.lakka.tv/doc/Configuring-Lakka/) ? You did not use the [GUI](http://www.lakka.tv/doc/Audio-settings/#switching-audio-device-from-the-gui) ?

Can you try to use another distro and check if the sound is working correctly with RA but you will have to remove PulseAudio to make the test.

ToKe79, 2019-06-26:
hi,
we have new [alpha images based on LibreELEC 9](http://nightly.builds.lakka.tv/Lakka-LE9.0-experimental/). you can test with these. you may also try to change `audio_rate` to `44100` via cmd-line:
```
# systemctl stop retroarch
# sed -i /storage/.config/retroarch/retroarch.cfg -e "s/^audio_out_rate = \"48000\"$/audio_out_rate = \"44100\"/"
# systemctl start retroarch
```
or via menu / GUI: Settings -> Audio Settings -> Audio Output Rate (kHz).

ToKe79, 2019-07-16:
we dropped LE9.0 base and moved to LE9.2, so new URL for images is http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/

TMueller83, 2019-07-16:
Thank you very much for the info @ToKe79. I can confirm that the HDMI audio output works fine now on my RX 580 using the LE9.2 image. A big THANKS to the complete Lakka development Team!

closed 2019-07-16 by TMueller83