libretro/Lakka-LibreELEC #638: NOOBS tarballs are v2.0 instead of v2.2

matthuisman, 2019-04-09:
According to http://mirror.lakka.tv/noobs/Lakka_RPi2/os.json the NOOBS version of Lakka should be 2.2.

However, I just installed and it’s older (v2.0)

Build Date: April 24 2017
Git version: 2a13af5

No Ethernet or WiFi working (using Pi3B+) which is another sign of older build.

I’m thinking someone updated the os.json but forgot to update the actual tarballs.

I downloaded the tarball and checked the os-release file. VERSION=“2.0” not 2.2

http://mirror.lakka.tv/noobs/Lakka_RPi2/
http://mirror.lakka.tv/noobs/Lakka_RPi/
(You can see the modified times have been updated for os.json etc, but not the tarballs)

Also posted on the forums:
https://forums.libretro.com/t/lakka-noobs-tarballs-are-v-2-0-instead-of-v2-2/21741/2

Thanks,

Matt

gouchi, 2019-05-04:
Hi,

We have updated the package manually, can you confirm it is working correctly now ?

Thank you.

matthuisman, 2019-05-05:
Perfect.
That has sorted it.
Thank you,

closed 2019-05-05 by matthuisman