libretro/Lakka-LibreELEC #641: Lakka_disk empty

xslackx, 2019-04-25:
Something is wrong with the image Lakka-RPi2.arm-2.2.2.img
This morning I wrote the SD card using the dd command, but at the end of the process the partition LAKKA_DISK was empty, it only contained the lost and found folder and a hidden file.
I have used raspberry pi b + v1.2.
Thank you

natinusala, 2019-04-25:
Isn't LAKKA_DISK the storage partition? If so, it's normal to be empty, you need to boot at least once for it to be resized and populated

xslackx, 2019-04-25:
This procedure has already been performed.
I left for more than 10 minutes connected but nothing happened, the LEDs turn on normally, I downloaded and burned the sdcard again and the same error occurred.
The card is working normally because I tested another operating system on the same sdcard.

ToKe79, 2019-04-25:
the case might be it is not booting? are you using correct image? is your sbc rpi0/rpi/rpi2/rpi3?
try RPi.arm instead of RPi2.arm?

xslackx, 2019-04-25:
My board is raspberry pi b + v1.2
I already tried to use the image of raspberry pi 3 and also raspberry pi 2 and both presented the same error.
I will try to use the image from the first version.

natinusala, 2019-04-25:
Are you using the RPi2 Image on a RPI1? That won't work

xslackx, 2019-04-25:
No, I'm using the rpi2 image and it still throws the error.
md5sum
38faf8cf23a0384a953fc428e39d7e6a Lakka-RPi2.arm-2.2.2.img.gz
I took a print
https://imgur.com/pr787Vq

natinusala, 2019-04-25:
> My board is raspberry pi b + v1.2

So is it a RPi1 or RPi2 ?

xslackx, 2019-04-25:
Eu desisto.

closed 2019-04-25 by xslackx