libretro/Lakka-LibreELEC #643: Lakka failing to load cores because of missing /dev/snd/seq`

mikesparky, 2019-05-12:
I'm using Lakka 2.2.2 x64 version on the x64 PC. The PC is a Lenovo Thinkcentre m93p, which employs i5-4570T with the integrated Intel Video.

The problem is that some Cores are not working: I've found out that GameCube (Dolphin) and PSP (PPSSPP) cores are failing to load, the Lakka is restarting when I try to load any their content.
In both cases in the logs there is a suspicious line:

```
open /dev/snd/seq failed: No such file or directory
```
This file is indeed not present on the lakka.

The audio is present in some other cores (NES, SNES, Genesis), so it's not a hardware problem. Also, if I enable the advanced settings in the GUI, I can see that the audio driver which is selected is `alsathread`.

natinusala, 2019-05-12:
I believe that this log line is about missing MIDI driver, it's not relevant to your error

mikesparky, 2019-05-12:
Thanks for pointing this out. Indeed, the last line is Segmentation Fault. I include the log and also change the topic.

[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/3170171/log.txt)

mikesparky, 2019-05-12:
Also, enabling Hardware Shared Context in the advanced core settings does not help.

mikesparky, 2019-05-12:
Also, I checked on the ubuntu live-cd and the ppsspp runs fine with the opengl backend, so the issue is not the hardware.

mikesparky, 2019-05-13:
I run the retroarch with gdb, however it cannot find any debug symbols for the lib ppsspp_libretro.so Can I find somewhere the debug version of this library so I can trace the error?

natinusala, 2019-05-13:
You need to rebuild Lakka with `DEBUG=yes` to have debug symbols

Nousjadeul-Ger, 2019-05-21:
I have the same Lenovo M93P setup and I'm experiencing issues with cores not loading as well.  Dreamcast and GameCube.  They have worked with previous versions of Lakka but the latest is not functioning.  There are issues with sound not playing on certain cores as well.  But my primary concern is with the GameCube and Dreamcast cores no longer loading.  I tried the latest nightly build (20190516) and I'm still experiencing the same issue.  What can I provide to assist with resolving this issue?

natinusala, 2019-05-22:
What I would do is make a debug build with the crashing cores updated, and with debug symbols. This way we would see if updating the core fixes the crash, and if not, what causes it.

@kivutar do we have enough space on the build server? can I make such an image? or @gouchi if you already have one ready?

Nousjadeul-Ger, 2019-05-22:
I'm happy to help but I'm not familiar enough with the debug build process.  If you can provide me with an image I'm willing to install it and pull the logs.

Currently I have lakka installed on the internal HDD and the roms are on the same HDD.  Is there any risk of roms being deleted if I install the debug build and then go back to the normal build?

ToKe79, 2019-05-23:
@natinusala I modified the nightlies job to include the debug flags and started new builds for all targets. They will start appearing at http://nightly.builds.lakka.tv/2019-05-23/ once they are finished (2-3 hrs for each target/device, as I could not reuse the existing build folders - the DEBUG flag is just not enough to trigger the recompilation of a package).

@Nousjadeul-Ger I am afraid if you install to the HDD again, it will be wiped before the install. Just use the USB live boot method or backup your roms before installation.

natinusala, 2019-05-23:
Okay, nice! Be careful though, the debug flag for the whole system doubles the size of the SYSTEM squashfs image, and it will probably be too large to fit in the FAT32 partition

ToKe79, 2019-05-23:
Indeed, the fs size is above 700 Mbytes (for i386). I increased the SYSTEM_SIZE (distributions/Lakka/options) to 768 (Mbytes).

ToKe79, 2019-05-23:
Huh, for x86_64 the 768 Mbytes were not enough, so I created image with 1 Gbyte boot-partition (overkill, I know, probably extra 10-100 Mbytes would be enough). Anyway, I cancelled remaining DEBUG builds. Debug build for x86_64 can be found here: http://nightly.builds.lakka.tv/2019-05-23/Generic.x86_64/

Nousjadeul-Ger, 2019-05-30:
I installed the 5/23 build on a flash drive and I'm able to boot into it.  But I'm having the hardest time connecting via SSH with Putty on windows 10.  I can ping the lakka IP address and get a connection but I get a connection refused error every time I try to connect with putty.  Any ideas?

To add I've added a few Dreamcast and Gamecube games and I get system freezes each time.  With the Gamecube I tried Crash Nitro Kart (which I've been able to load before) and it just freezes on the intro screen while the game is loading.  In yellow it states at the top "Video Info: Intel Open Source Technology Center, Mesa DRI Intel(R) Ha..." and it cuts off.  With Dreamcast I just get a black screen and nothing else.  I have to press the power button each time to get out of it and shut down.

I know that a log file would be infinitely better, I just need some help accessing it.

ToKe79, 2019-05-30:
Did you enable SSH in the Settings -> Services menu? If so, probably the SSH server key changed (?) and PuTTY refuses to connect because of false alarm (MITM attack), but in such case you would get more verbose "can't connect / fingerprint changed / are you sure" error/warning than just plain "connection refused". I suspect the SSH service is not running.

Nousjadeul-Ger, 2019-05-30:
I did and confirmed that it was on before accessing the ethernet IP address from Information.  I also enabled it during boot by typing in 'live SSH'.  I'm at a bit of a loss at this point.

mikesparky, 2019-06-03:
I tried the Lakka-Generic.x86_64-2.2-devel-20190523154650-r28266-gd57fdb3.img.gz and the GameCube emulator core started working. Will check the PSP later though.

mikesparky, 2019-06-03:
The PPSSPP core is also working. There were some other minor bugs, but that's how it happens. Can we know when the next release is scheduled?

gouchi, 2019-06-11:
For the Midi issue, we will have to [enable it](https://github.com/libretro/Lakka-LibreELEC/issues/628#issuecomment-480612586) as some cores need it.

closed 2020-08-18 by Ntemis