libretro/Lakka-LibreELEC #644: Preinstall unrar, 7z, xz etc..

snbuback, 2019-05-19:
Lakka version 2.2

Frequently roms pack are rar/7z/xz,etc.. files. Systems like raspberry are slow to upload individual files so is better upload the compressed ones, but there is no too to unpack.
These tools are smaller and are already part of the package system in debian for example.

The idea is able to run on the command line. I'm planning to have them on the interface.

RobLoach, 2019-05-19:
.zip is supported. Other compression methods are iffy. Doesn't work for CD media.

snbuback, 2019-05-20:
I'm aware of the zip. But many rom packs are rar or 7z (few are zip due to the size and low compression rate)

RobLoach, 2019-05-21:
Would need to be added to RetroArch before in Lakka. Here's an issue for ya! https://github.com/libretro/RetroArch/issues/4122

closed 2019-05-21 by RobLoach