libretro/Lakka-LibreELEC #646: Please use ExFAT for majority of partition

jeapostrophe, 2019-05-20:
This is not a bug, but a feature request.

This page talks about how to access the Lakka drive --- https://github.com/libretro/Lakka-LibreELEC/wiki/Accessing-Lakka-filesystem#direct-drive-access --- but I think it would be easier for users if the data partition were just already in a format like ExFat that is a standard for SD cards.

I use Lakka with a Pi3B+. I would love if it were easier to access, given that the network and drive access are so slow using the Pi hardware.

natinusala, 2019-05-20:
What does exFAT have that ext4 doesn't?

jeapostrophe, 2019-05-20:
Read/write support for free on all platforms without poor performance of
FAT32

natinusala, 2019-05-20:
The thing is, a Linux system can't boot off of an exFAT filesystem. Lakka Switch does it for FAT32 but it requires countless hacks due to the lack of symbolic links, the invalid `:` characters, etc etc

jeapostrophe, 2019-05-20:
It doesn't need to boot off of it. It could just have a separate boot partition perfectly sized (given that we can compute how much space is needed) and then everything else as a data partition

natinusala, 2019-05-20:
Again most services like bluetooth and wifi rely on symbolic links for their configuration. Symbolic links don't work in exFAT so they need to be removed.

Bluetooth uses `:` in its data files, which is invalid in exFAT. Pulseaudio uses symbolic links too.

overlayfs, a very important component of Lakka, doesn't work in exFAT, it needs to be replaced by mergerfs.

Mr-Jake, 2019-05-28:
This is not a request, but is it possible to create an extra exFAT partition on the SD card that can hold only roms. I was actually thinking of using partitioning software to modify the partitions once the SD card is created. 

Seems most users would like to just drag and drop roms to their SD card from a Windows machine.

natinusala, 2019-05-28:
What kind of ROMs would you put in an exfat partition that you can't already put on a fat32 partition?

Mr-Jake, 2019-05-29:
My mistake. I was thinking the storage partition was still ext4. It did not realize it was recently changed to fat32. Apologies.

jeapostrophe, 2019-05-29:
I interpret @natinusala question as saying "ExFAT requires installing new code, but we already support FAT32, so we could do what you want more easily if it FAT32". But @Mr-Jake is inferring that the partition already exists. ExFAT's advantage is that it is faster and supports bigger files, but is probably not much better than FAT32.

natinusala, 2019-05-30:
No I was literally asking what is the point of using exFAT, we currently have zero needs for files over 4gb

ToKe79, 2019-05-31:
AFAIK, for the second/data partition, which is mounted to `/storage`, ext4fs is used. It is same for LibreELEC (on which Lakka is based):

https://github.com/LibreELEC/LibreELEC.tv/blob/f6370fe64fda9b1fe319a2d0c330980e2bd35bea/scripts/mkimage#L257-L270

Some of the reason (why ext4) were already mentioned:
- overlayfs
- colon (`:`) in file names (Bluetooth saves configs based on the MAC address of the device, config files are stored in the $HOME of the user root, i. e.  `/storage/.config/...`)
- symlinks

FAT32 is used for the first/boot partition on which the kernel and system/rootfs (readonly squashfs in a single file) reside.

But you are free to repartition your media and add ExFAT partition for your needs and mount it to `/storage/roms`, or use separate (USB) disk formatted to the FS of your choice (of course you are limited to those supported by Lakka), which will be mounted automagically to `/storage/roms/<LABEL>`.

Fun fact: The FAT32 (boot) partition is mounted to /flash (as read-only by default) and can be remounted to read-write:

     mount -o remount,rw /flash

Ntemis, 2019-06-14:
is this still need to stay open?

closed 2019-06-14 by natinusala

DK189, 2020-08-11:
Can load rom from exFat USB?? exFat can easy read-write on mac os, windows, linux (?)...

natinusala, 2020-08-11:
Try it and see

Danixu, 2021-03-07:
I think that one of the best things in Recalbox is its exFAT data partition... I'm able to copy games at about 70MB/s to my SD card, instead the never ending copy at 1MB/s using the Wifi connection (i'm using a RaspberryPi Zero).

natinusala, 2021-03-07:
You know that you can use USB with Lakka too right

Danixu, 2021-03-07:
Yes, but I'm using Lakka on a Retroflag GPi and connecting anything by USB is a bit hard and dangerous (the USB can be broken).

rinzwind5, 2023-02-11:
"Accessing ext4 partitions from windows is not supported natively but you can install a [driver](https://www.howtogeek.com/112888/3-ways-to-access-your-linux-partitions-from-windows/)."

The replies here are not helpful and ignore Windows compatibility. It is obvious we want at least an option to format the rom storage folder as exfat so file transfers using Wiindows can be fast. Wifi transfer on a pi is a pain, ext4 on Windows is a pain. Reality check: most people use Windows.

ToKe79, 2023-02-11:
Just put your roms on extra USB thumb drive formatted with ExFAT using Windows and stick it to a free USB port on your Lakka device. There are objective reasons for using ext4 already explained. Yes, there is a possibility to add additional partition (i.e. reduce the ext4 to a smaller size - ~4G or less) and use the remaining space for FAT32/ExFAT partition. This could be done during the first boot, but would introduce incompatibility to older installations. For now it still remains a nice to have, as there is very easy workaround (i.e. using the aforementioned method with additional USB drive with roms).

Also using separate drive for your roms makes it easier in case of full reinstall of the system. Any additional storage partition connected to Lakka will be mounted in its own subfolder under `/storage/roms/<PART_NAME>/`.

Danixu, 2023-02-11:
As I said in my other comment, when I asked for this my intention was to use it on a GPi case, which uses a RPi zero. This RPi is quite slow in wifi access, and to have an external thumb is dangerous (the connector is exposed and can be broken), reduces the autonomy, and forces me to buy an USB connector for the case.
About the windows compatibility layers for ext4, all of them are a piece of sh*t. I've even used a paid one expecting to have a better compatibility than the free open source version, and all I've got is the possibility to damage the ext4 partition and to have to format it again losing all.

All the solutions proposed are very bad, but luckily there are alternatives and I'm using one of them right now.
Also I can try to reduce the ext4 partition using Linux, format the new space to ExFAT, and modify the mount points to use it as roms storage keeping the problematic one in the ext4 partition, but I wanted a solution out of the box because if not the update process can be very painful.
You talk about the compatibility but this can be easily fixed just with an script to rename the folders when old versions are detected, which is the main problem of the migration from ext4 to ExFAT.

Anyway, my intention was to simplify the life to all the Windows users (and also a bit mine because my main gaming laptop is using Windows), because I have knowledge of Linux and two laptops with it (I work every day with it), and to rid with this is not a big problem for me.

Ntemis, 2023-02-11:
Exfat doesn't support symlinks . Blame Microsoft ¯\_(ツ)_/¯
Double close

ToKe79, 2023-02-11:
Exactly - easy and dirty solution is to shrink the ext4 partition and add new fat32/exfat partition. This can be done during the first boot, but:

- the exfat partition cannot be used for merging with the "supplied" file system. for example - system folder (with bios files) also includes some files required by many cores (such as assets, etc.) - this means the system ("bioses") folder cannot be "moved" to the exfat partition, as some cores would be missing these files / it is up the the user to make sure these files are placed here after they decide to remap the system folder to a new location. same logic applies to folders like joypads, databases etc (that are "merged" under /tmp/xxx during runtime)
- configuration files / services settings will be still saved to the folders on the ext4 partition, which still can be accessed via smb/ssh

I will prepare POC and test, if this new partition will be automounted under /storage/roms by udevil, or an additional mount unit is required.

rinzwind5, 2023-02-12:
> What does exFAT have that ext4 doesn't?

Perrrfect Windows SD card copy/paste support DUH ; which you try hard to ignore because possibly you hate Windows and everything with it and thereby ignoring the purpose of this whole thread? Which hurts this great project, a workable out-of-the-box experience for us poor Windows users (the majority). Yes, a lot just buy a preconfigured device with lots of illegal roms preloaded. But the rest wiith a clean pi... which is a major use-case. Think it over. Thank you.

Danixu, 2023-02-12:
> Exfat doesn't support symlinks . Blame Microsoft ¯_(ツ)_/¯ Double close

And what is the problem?, roms, configurations, saves... I'm pretty sure that if not all, most of them don't need symlinks. This is a good feature, but looks like you have any kind of problem with Microsoft products and you don't want to even consider it. I don't like Windows, but is what we have if we want to play games and is the most used SO, so too simplify the life to all that persons which have to use it is not a bad idea.

ToKe79, 2023-02-12:
@Danixu @rinzwind5 no need for such comments, nobody expressed any hate towards Microsoft or Windows.

as mentioned, we cannot drop the ext4 partition, but I am doing last tests with **additional** exFAT partition.

Danixu, 2023-02-12:
@ToKe79 Sorry, but considering how useful can be and the reasons used to say no... I can only think about any hate against Microsoft, avoiding the use any of their filesystems at any cost. We know that ExFAT is far away from perfection, but it's the only filesystem that is compatible with all platforms.
Try to copy 300gb of roms through wifi and you will die trying it.

ToKe79, 2023-02-12:
@Danixu I have finished the changes needed for optional exFAT partition. I am preparing test images for some devices, including both GPICase (zero (w) and zero 2w) images from 4.x branch. As soon as these are ready, they will be available at https://nightly.builds.lakka.tv/members/vudiq/exfat-test/

It is not possible to add exFAT partition to already existing installation (i.e. where the whole space is already taken by the ext4 partition) - it is too risky for automated script. However for any new "installation" you can add exFAT partition following these steps:
1) download the image
2) flash the image (as I am on Windows, the most reliable and user friendly image writer by my opinion is the Raspberry Pi Imager - no huge electron app, you can even use it to flash Lakka images for various devices using the built-in navigation)
3) re-plug the storage device (USB drive / SD card) to trigger windows mounting the disk
4) navigate to the LAKKA partition
5) rename `no_create_exfat` to `create_exfat` (if there is no `no_create_exfat` then just create empty file called `create_exfat`; the reason for `no_create_exfat` is that on some systems it might require more steps to create a file without an extension, so there is already one file without extension, that can be just renamed; the file can be safely removed after first boot / when the partition setup is done, but there is no harm in leaving the file there, as the file is checked only during partition setup)
6) boot the USB drive / SD card in your device - you should see the progress of partition setup followed by a reboot. when you get into RetroArch, you can check `Import Content` - `Scan Directory` - `EXFAT` folder is the mount point for the exFAT partition (full path `/storage/roms/EXFAT`).

The ext4 partition will occupy ~2GB and the exFAT partition will take over the remaining space.

BIOS files: You can use the exFAT partition to transfer bios files - using your computer create `bios` folder in the root of the exFAT partition and place your files there. Place the disk into your device and boot - all bios files will be copied to the default location expected by RetroArch and backed up to `bios_copied` folder on the exFAT partition, which you can remove. In this folder also a transfer log will be saved, so you can check for any errors.