libretro/Lakka-LibreELEC #647: XU4 - Controllers are randomly assigned to USB ports at reboot

ttmcmurry, 2019-05-22:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
**2.2.2**
- What system hardware are you using?
**XU4**
- What did you do?
**XU4 has a 7-port USB 2.0 hub attached to the USB2 port.  USBHUB/PORT1 has an 8bitdo Retro Receiver (RR) on it, USBHUB/PORT2 also has a RR on it.  Boot the XU4.**
- What did you expect to happen?
**Controller 0 should map to the 1st USB port, CONT1 -> USB2, and so on**
- What happened instead?
**Random Controller-to-USB-Port assignments**

The original goal here was to have 4 simultaneous controllers on the XU4, which only has a total of 3 USB ports, to play 4-player Micro Mages on NES .. and probably some other games, too.  What ultimately ends up happening is either LibreELEC or Retroarch (and I'm leaning toward LibreELEC as the culprit) is assigning the first responding USB controller on any USB bus as "Controller 0".  The next one is Controller 1, and so on.  

The only workaround I've found is to unplug all RRs, then plug them back in, one at a time into the hub.  Until I shutdown/reboot the XU4, everything works fine.  Unfortunately the mappings do not persist between reboots.  Lastly, even if only one RR is plugged in, sometimes it gets assigned a "non-controller-0" position.  Once I remove and add the RR back to the Hub's 1st USB port, gets assigned to Controller 0 and works.  The workaround is 100% consistent.

If I had to take a guess at what's happening, is some portion of Lakka isn't considering external USB hubs in the USB device tree, and so when Retroarch claims a controller to a device mapping, it is assigning the controller to whatever the 1st device is, as presented from the XU4's root USB hub, not the actual (external) root hub device.  It's as if 7 devices are presented with the same ID to Retroarch so it picks the first one it sees, without evaluating the full USB device tree for proper controller assignment.

ttmcmurry, 2019-05-23:
Adding dmesg and lsusb logs.  

Of Note:

- At first boot, the wrong driver is installed when the RR is detected  USBHID vs XPAD.  I think the driver issue is more at how the RR is designed and it defaults to an "Auto" mode, which switches the driver to whatever the last connected device was using.  If I power on my Xbox One controllers prior to turning on Lakka (this is a powered USB hub), then the correct driver is always present.  May be worth a note on using 8bitdo's Retro Receivers in the documentation.  Turn on the controller first before Lakka.
- Otherwise removal & reinsertion of the RR chooses the correct XPAD driver, if either Lakka was running first, or the controller was connected prior to boot.  This would be easier if RRs would just remember the last mode they were set to.
- I wonder if, at Lakka's boot, it should just issue a USB reset command to all *connected* root hubs.  For example, on the XU4, the Ethernet port is connected to the Exynos USB3 root hub, probably wouldn't want to reset that.  This way, when a powered USB hub is connected, the OS has booted far enough along for a connected device to properly detect what it's connected to.  This is assuming a reset isn't already happening, or at least happening late enough in the game for the Retro Receiver to accurately detect what it's plugged into.
- Controller assignments continue to be random regardless of using ports 1-3; or the subports of 4 (4-8).
- At one point, the external USB hub changed bus numbers from BUS1 to BUS2.  Seems like LibreELEC may be assigning buses to root hubs at random. 
- This 7-port hub in my example is designed with an additional root hub inside the external hub's root hub.  Physical Ports 1-4 are cascaded behind port 4.  Physical Ports 5-7 are in the root hub.  Port 8 is the uplink to the exynos root hub. I have no idea if this has any bearing on the randomness of the controller assignment.  It's always random.

Fully populated External USB Hub:

/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=exynos-ehci/3p, 480M
    |__ Port 1: Dev 31, If 0, Class=Hub, Driver=hub/4p, 480M (Rear Uplink) 
        |__ Port 1: Dev 35, If 3, Class=Vendor Specific Class, Driver=, 12M (Port 7)
        |__ Port 1: Dev 35, If 1, Class=Vendor Specific Class, Driver=, 12M (Port 7)
        |__ Port 1: Dev 35, If 2, Class=Vendor Specific Class, Driver=, 12M (Port 7)
        |__ Port 1: Dev 35, If 0, Class=Vendor Specific Class, Driver=xpad, 12M (Port 7)
        |__ Port 2: Dev 36, If 0, Class=Vendor Specific Class, Driver=xpad, 12M (Port 6)
        |__ Port 2: Dev 36, If 3, Class=Vendor Specific Class, Driver=, 12M (Port 6)
        |__ Port 2: Dev 36, If 1, Class=Vendor Specific Class, Driver=, 12M (Port 6)
        |__ Port 2: Dev 36, If 2, Class=Vendor Specific Class, Driver=, 12M (Port 6)
        |__ Port 3: Dev 37, If 0, Class=Mass Storage, Driver=usb-storage, 480M (Port 5)
        |__ Port 4: Dev 33, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 4: Dev 41, If 0, Class=Mass Storage, Driver=usb-storage, 480M (Port 4)
            |__ Port 2: Dev 39, If 0, Class=Mass Storage, Driver=usb-storage, 480M (Port 3)
            |__ Port 3: Dev 40, If 0, Class=Mass Storage, Driver=usb-storage, 480M (Port 2)
            |__ Port 1: Dev 38, If 0, Class=Mass Storage, Driver=usb-storage, 480M (Port 1)

[USB Troubleshooting.txt](https://github.com/libretro/Lakka-LibreELEC/files/3210440/USB.Troubleshooting.txt)

Ntemis, 2019-06-12:
Please test this image thanks
https://www.androidfilehost.com/?fid=6006931924117893548

ttmcmurry, 2019-06-12:
I'll give this a try later today/tomorrow when time allows.  Thank you!

ttmcmurry, 2019-06-12:
Unfortunately the image doesn't boot properly.  See image below.  Tried re-flashing and rebooting but this is all it does.  I tried accessing it via SSH, but PuTTY times out.  

[image]

Ntemis, 2019-06-13:
Please test this new image too
https://www.androidfilehost.com/?fid=6006931924117894778
 Thank you for your time
Burn with etcher please and verify because previous image is bootable too

ttmcmurry, 2019-06-13:
That 2nd image works, it also has a different partition layout than before.

There are two effects I'm noticing.

1)  The physical ports on this hub are numbered opposite their logical assignment.  I would imagine that not all USB hubs are going to always be made the same, and sometimes they will and sometimes they won't match:
```
Physical Port 8 = B1-P1-D2       (this is the USB Hub Uplink Port)
Physical Port 7 = B1-P1-P1-D8 
Physical Port 6 = B1-P1-P2-D9
Physical Port 5 = B1-P1-P3-D10
Physical Port 4 = B1-P1-P4-D4-P1-D11
Physical Port 3 = B1-P1-P4-D4-P2-D12
Physical Port 2 = B1-P1-P4-D4-P3-D13
Physical Port 1 = B1-P1-P4-D4-P4-D14
```
2)  As long as I insert the Retro Receivers from left-to-right, they seem to stay in sequence now, survive reboots, and shutdowns.  

I want to give it some more time, but I have to step away for a bit.  Had some trouble with the controllers working between shutdowns, and wanted to make sure it was either me (likely) or the retro receiver defaulting back to "nintendo switch (auto)" mode (also likely).  

lsusb tree:
Lakka:~ # lsusb -t
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/2p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/2p, 480M
        |__ Port 1: Dev 3, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 2: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 4, If 2, Class=Human Interface Device, Driver=usbhid, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=exynos-ohci/3p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=exynos-ehci/3p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 6, If 0, Class=Vendor Specific Class, Driver=xpad, 12M
        |__ Port 1: Dev 6, If 1, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 1: Dev 6, If 2, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 1: Dev 6, If 3, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 2: Dev 7, If 0, Class=Vendor Specific Class, Driver=xpad, 12M
        |__ Port 2: Dev 7, If 3, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 2: Dev 7, If 1, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 2: Dev 7, If 2, Class=Vendor Specific Class, Driver=, 12M
        |__ Port 4: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M

Ntemis, 2019-06-14:
So can we close this down as resolved?

closed 2019-06-14 by Ntemis

ttmcmurry, 2019-06-14:
There are 2 concerns that I see:

1)  Documentation - It would be worth mentioning in Lakka docs that due to the varying nature of USB and USB Hub design, the USB ports may be reverse from what they are labelled.  If your controllers are reversed after reboot, then reverse the order they are plugged in.  

2)  When > 1 controller is plugged in via Retro Receiver, until all controllers are online, RetroArch does **not** recognize the controller input.  Only after the last controller is online and in XINPUT mode and a button pressed on the last ordered controller, does Retroarch respond to _any_ controller input.  Scenario 2 did not occur before now, this may be a new issue and require a new PR.

Ntemis, 2019-06-14:
@twinaphex do you have any thoughts about these concerns from @ttmcmurry ?