libretro/Lakka-LibreELEC #652: Lakka 2.2.2 Performance issue

simply1233, 2019-06-03:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

Ntemis, 2019-06-14:
we cant help you by only this info. sorry.
Make sure you have a decent heatsink with a fan on it and report back if issue persist with logs.
Closing it down atm, let me know

closed 2019-06-14 by Ntemis

simply1233, 2019-06-19:
I already had a heatsink, I also had a fan.