libretro/Lakka-LibreELEC #655: Controls aren't remappable in prBoom

brightentayle, 2019-06-05:
The prBoom core shipped with Lakka 2.2.2 for the Raspberry Pi 2/3 seems to be stuck on default gamepad controls no matter how I remap said controls in Retroarch. For example, when I remap Fire from X (top) to B (bottom) and save the settings, it appears to be binded to B in the Controls menu, but during the gameplay, prBoom still uses X instead.

RobLoach, 2019-06-05:
Does this happen with any other cores? Might wanna make an issue over at the core issue queue:
https://github.com/libretro/libretro-prboom

brightentayle, 2019-06-06:
I was pretty sure controls were remappable in the prBoom core that was compiled on Retropie from source just yesterday. Though it turns out it's not a core-related issue at all - my copy of Lakka doesn't let remaps work as intended, even though they are saved, regardless of the core picked.

Not exactly sure what causes this. Might try a fresh install with a different SD card to make sure it happens consistently.

brightentayle, 2019-06-06:
Apparently, the bug doesn't occur on a different SD card. It's probably an anomaly, though I'm still not sure what causes it to happen exactly.

MicrosoftTakeover, 2019-12-13:
Can we change the PrBoom default controls shipped with Lakka? It's a bit hard to play on controller out of the box, Quake's default controls are much better in this respect.

closed 2020-08-18 by Ntemis