libretro/Lakka-LibreELEC #662: Lakka 2.2.2 Reicast Crash

iokzero, 2019-06-08:
Lakka 2.2.2 - 64Bit on 7 Gen Intel NUC.
Launch anything with Reicast crash the system.
Launch Reicast, game runs.

natinusala, 2019-06-08:
>Launch anything with Reicast crash the system.
>Launch Reicast, game runs.
So, does it run or does it crash?

Could you please provide more details, maybe logs of the crash too?

iokzero, 2019-06-08:
Sorry about the report:, try to follow the report guideline but forget to label things. 

So I did a retroarch  -v
when It launch, it load many stuff, then at the end it just  **segfaults**  at the end. 
retroarch will then reload itself。

Tried nightly build, it got its own issue, I can't even get it to connect to ssh.

natinusala, 2019-06-09:
I started a debug build to send you, don't hesitate to remind me if I forget it in a corner

gouchi, 2019-06-09:
Indeed I reproduce the segfault using nightly devel-20190608205405-r28277-g8387e1bd1 on Generic x86_64
```
SIGSEGV @ 0x7fdfb58aca87 ... 0x16 -> was not in vram (dyna code 0)
[libretro INFO] Fatal error : segfault
 in signal_handler -> core/libretro/common.cpp : 398 
DEBUGBREAK!
SIGSEGV @ 0x7fdfb583b390 ... 0x7fdfb583b390 -> was not in vram (dyna code 0)
[libretro INFO] Fatal error : segfault
 in signal_handler -> core/libretro/common.cpp : 398 
DEBUGBREAK!
Illegal instruction (core dumped)
```
We will have to check it with debug builds.

iokzero, 2019-06-09:
Thank you so much for the help, 
I have a SFF X86_64 PC I could use for beta testing, and I would love to help debug the X86(32bit/64bit) whenever I can in the future too. Definitely an awesome project you are doing.

natinusala, 2019-06-10:
Here is the debug image https://transfer.sh/mNjsm/Lakka-Generic.x86_64-2.2-devel-20190610112753-r28277-g8387e1b.img.gz

@gouchi can you try to see what's wrong with it? or should I give instructions to @iokzero ?

iokzero, 2019-06-10:
I would like to know the instructions too if you don't mind.

natinusala, 2019-06-10:
Basically you need to enter SSH, stop retroarch with `systemctl stop retroarch` then do `gdb retroarch`

Inside the GDB prompt you need to do `start` to start RA and `continue` to go past the first breakpoint. Now make it crash by running a game in reicast. Give us the output of `bt full` once RA has crashed.

mlnlbrt, 2019-06-10:
Hi guys, take a look at this:
https://github.com/libretro/Lakka-LibreELEC/pull/632/files

I bet it will fix the issue.

natinusala, 2019-06-10:
Hmm, I really need to go through all PRs and merge / close them lol

@iokzero here is a build with the above fix included, does it work for you? https://transfer.sh/lViua/Lakka-Generic.x86_64-2.2-devel-20190610181523-r28280-g7ca1eef.img.gz

iokzero, 2019-06-11:
@natinusala Have tested the image you listed, Reicast works as expected！

Thank you so much for the help！

FYI, it seems that the notification system have been rewritten, but the setting for notification have not, so most of the setting don't do anything at the moment, but I think that's RetroArch's issues, not lakka specific.

iokzero, 2019-06-11:
Just curious, is it possible to implement this back to the Lakka 2.2.2 ? 
Lakka 2.2.2 with RA 1.7.6 seems really stable for me, the current 1.7.7 seem have way too many incomplete features. 

Not trying to push extra works for others, just wondering if there's simple way I can implement this myself (or whoever have similar issue).

Many thanks

ToKe79, 2019-06-11:
I checked out d51e47b1ec0b3e30243bbab615e523b87975cdb2 (commit from March 12 - day before 2.2.2 was published - no tag found, so I picked this commit) and applied these commits:
- 8b22eefe5e85b2b820595009f5cc6f588ccc6615 (retroarch shutdown hack)
- f16e9ba48b2ccae8b4a8d2544849dd37fc628eb4 (various buildfixes)
- d668ac4c5ef820e227ecbcff5ca13ec16293e3ae (fix reicast segfault)
As soon as the image is ready I will post a link.

natinusala, 2019-06-11:
Thanks for testing!

ToKe79, 2019-06-11:
#632 relies on host mesa-dev and does not use the GL/gl.h et al provided by the sysroot build system. So the first build failed due to missing mesa headers on my build system. I think it should use the sysroot's mesa headers and not the ones on the host in /usr/include/..., am I right?

So new route was:
- checkout d51e47b1ec0b3e30243bbab615e523b87975cdb2 -b test_branch
- cherry-pick 8b22eefe5e85b2b820595009f5cc6f588ccc6615
- cherry-pick f16e9ba48b2ccae8b4a8d2544849dd37fc628eb4
- fetch upstream pull/658/head:flycast
- cherry-pick f2093e9aa27e3a6bb3c40082b853aa9836f53778
- resolve conflict in `distributions/Lakka/options` (leave changes of f2093e9, but remove `theodore` from LIBRETRO_CORES - this core was not included in 2.2.2)
- resolve conflict in `packages/libretro/core-info/package.mk` (leave changes of f2093e9)

As different mesa headers were used I am not sure if the segfault will be fixed.
Here is the image: http://nightly.builds.lakka.tv/special_builds/Lakka_2.2.2_Generic.x86_64_fixed_reicast/
Please, let us know, if this image works (the core name is now `flycast`, not `reicast`).

iokzero, 2019-06-11:
@ToKe79 

Thanks for the special built.

But the special_builds Did NOT fix reicast (flycast) crash issue.

mlnlbrt, 2019-06-11:
@ToKe79 patch from #632 has nothing to do with reicast build pulling GL/gl.h from host mesa-dev, so I don't understand your comment :)

ToKe79, 2019-06-11:
Thanks to @mlnlbrt I have prepared two images to test - one is compiled from the master branch (RA 1.7.7) and the other is with Lakka 2.2.2 (RA 1.7.6). The images are here: http://nightly.builds.lakka.tv/special_builds/reicast_segfault_fix/
If you can test both, that would be very helpful.

iokzero, 2019-06-12:
@ToKe79 

Did some basic test on both image, Reicast works on both!
I will do a deeper and more thoroughly test in the next few days.

Thank you much for the help. if there's anything you would like me to test specifically, please let me know.

closed 2019-06-12 by ToKe79

iokzero, 2019-06-12:
@ToKe79 

So I did a relatively comprehensive test, on the image based on Lakka 2.2.2 (RA 1.7.6) fixed. 
Everything seems work fine, except an old bugs #605 on Lakka 2.2.0 reappear (it was fixed on Lakka 2.2.2 stable)

as work around, when you using the ozone menu driver, whenever you set "show advance menu" off, you need to hit esc to restart RetroArch to save that setting before you hit the back button. if you push the back button, RetroArch will crash as described on Bug #605 

I just want to post this bug here for now, sense it is based on a special built. free feel to move it to the right place.

Master branch based on RA 1.7.7 do NOT have this issue.

PR #672 (2019-06-12): reicast: properly fix segfaults with Generic
Removing Dockerfile updates which should be no longer needed and restore previous version of reicast patch, hopefully fixes #662