libretro/Lakka-LibreELEC #671: Why doesn't Lakka use system's already-provided toolchain?

AnClark, 2019-06-11:
I'm building Lakka from source, but I discovered that Lakka's build system always fetch these toolchain's source code, then build them, regardless of that they can be easily installed through an `apt install XXX`.

For example: GNU Make, CMake, Bison, libtool, GCC, etc.

As you know, most of time in building Lakka is wasted on building those toolchains.

ToKe79, 2019-06-11:
Because you are cross-compiling. This is not a bug, but a feature of the sysroot build system.
Locally provided "build essentials" are used to compile the toolchain, so they are not wasted.

closed 2019-06-11 by ToKe79

AnClark, 2019-06-11:
No wonder! But I saw that Android's build system provided prebuilt toolchain binaries. For example, GCC for multiple platforms and ABIs.