libretro/Lakka-LibreELEC #674: Does lakka support nvidia Jetson nano?

chaogehen, 2019-06-12:
I love Lakka, I hope to run it on nvidia jetson nano, will it be compatible? Thanks!

GPU | 128-core Maxwell
CPU | Quad-core ARM A57 @ 1.43 GHz
Memory | 4 GB 64-bit LPDDR4 25.6 GB/s
Storage | microSD (not included)
Video Encode | 4K @ 30 \| 4x 1080p @ 30 \| 9x 720p @ 30 (H.264/H.265)
Video Decode | 4K @ 60 \| 2x 4K @ 30 \| 8x 1080p @ 30 \| 18x 720p @ 30 (H.264/H.265)
Camera | 1x MIPI CSI-2 DPHY lanes
Connectivity | Gigabit Ethernet, M.2 Key E
Display | HDMI 2.0 and eDP 1.4
USB | 4x USB 3.0, USB 2.0 Micro-B
Others | GPIO, I2C, I2S, SPI, UART
Mechanical | 100 mm x 80 mm x 29 mm

https://developer.nvidia.com/embedded/buy/jetson-nano-devkit
https://developer.nvidia.com/embedded/downloads#?search=Jetson%20Nano

natinusala, 2019-06-12:
As I already said on Discord, the Switch port is based on the Jetson Nano Ubuntu distribution (Linux4Tegra revision 32-something). One can easily take the Switch port, remove the Switch changes and config (to use the regular Nano ones instead) and it should just work. The bootloader will need changes too.

That being said, I'm not sure that I want to support it here, on the official Lakka repo. L4T is really weird (it uses 8 different git repositories for the kernel alone), and I had to introduce many breaking changes in the Switch port to make it work (and it's super, super dirty).

So, your best bet is currently rebuilding Lakka Switch, until we find a way to properly add L4T, in which case the Switch port + Jetson Nano will be officially supported here.

Ntemis, 2019-06-14:
Close it maybe?

natinusala, 2019-06-14:
Not sure, I would like to have it as an issue because Jetson Nano is really something that I want to eventually have for Lakka

Ntemis, 2019-06-14:
we all do but is nvidia willing to sent some boards over?

mrcmunir, 2019-07-26:
Hi I don't have much free time but I would be interested in providing support to support nvidia jetson nano based on a 4.9.140 kernel and actually are based l4t 32.2.0 branch.

https://nv-tegra.nvidia.com/gitweb/?p=device/hardware/nvidia/platform/t210/porg.git (For dts files )
https://nv-tegra.nvidia.com/gitweb/?p=3rdparty/u-boot.git (For u-boot)
https://nv-tegra.nvidia.com/gitweb/?p=linux-nvgpu.git   ( Nvgpu driver)
https://nv-tegra.nvidia.com/gitweb/?p=linux-4.9.git (Kernel 4.9)

natinusala, 2019-07-26:
Again, Lakka-Switch has everything needed to compile a L4T version of Lakka, with nvgpu.

1. Take Lakka Switch
2. Remove the Switch specific repositories (use nvidia's gitweb instead of switchroot) and various userland patches
3. Change DTS
4. Change/adapt bootloader
5. ???
6. Profit

ezar, 2019-12-05:
Nobody lnterested in make Nano port?

Ntemis, 2019-12-05:
It must be done by someone that owns it.
Feel free to try what @natinusala suggested

ezar, 2019-12-06:
> Hi I don't have much free time but I would be interested in providing support to support nvidia jetson nano based on a 4.9.140 kernel and actually are based l4t 32.2.0 branch. [...]

@mrcmunir Can you try it?

ezar, 2020-01-02:
https://www.reddit.com/r/SBCGaming/comments/eia0yh/nvidia_jetson_nano_rotn_retropie_on_the_nano_v1/?utm_source=amp&utm_medium=&utm_content=post_body

mrcmunir, 2020-01-02:
@ezar Sorry For now I have not tried anything to much with lakka all the work is focused on Retro-Arena (At the moment internally)
Some system need debugging buit it's working 32 system and some platform experimental to do in comming

ezar, 2020-01-02:
thanks mrcmunir
i try Retro-Arena beta at jetson :) works fantastic

natinusala, 2020-01-03:
Someone on the switchroot team is willing to redo the Switch port on the LE9 branch, which will bring cleaner Switch and Jetson Nano support for both upstream Lakka and LibreELEC

closed 2020-08-18 by Ntemis